prusa3d/Prusa-Firmware #4370: [BUG] Noctua fan will not start spinning unless manually spun first

malhovic, 2023-09-05:
**Printer type** - MK3S+
**Printer firmware version** -  3.13.0 Revo

**MMU upgrade** - No
**MMU upgrade firmware version** - N/A

**SD card or USB/Octoprint**
  SD

**Describe the bug**
  Installed Noctua fan, replacing the Delta fan. Went through the full factory reset procedure and it will not "start" the fan. You can see it try but fail to spin the fan to which it throws a fan wiring error. If you spin the fan with your finger when you see it try to start the fan, it will then spin the fan and you can go into the menu's to read the RPM of the fan. The fan will continue to run and no error will be reported. I've attempted disabling altfan detection as well and that doesn't resolve the issue either.
  
**To Reproduce**
  Have the delta fan installed, install the Noctua fan, factory reset the printer with 3.9 or 3.13 and the fan won't start spinning
  
**Expected behavior**
  The fan should start spinning on it's own

cozmo14047, 2023-10-13:
I don't have this problem but have found that with these fans if something is stopping it from moving on first power then it won't move until you spin it this sounds more hardware then firmware

malhovic, 2023-10-13:
Did more investigation, the listing I purchased for the fan sent me a 12v fan when I ordered a 5v. I got a replacement and all is well now. This isn't a bug, it was hardware.

closed 2023-12-13 by 3d-gussner