prusa3d/Prusa-Firmware #4378: [BUG]3.13.1/3.0.0 MK3+/MMU2S Safety Timer firing during initial Setup causes weird printer state

trevorbattle, 2023-09-08:
MK3+, originally started as MK2 but upgraded along. Firmware 3.13.1 from github, compiled with VSCode windows.

MMU2S with 3.0.0 firmware. (Printer source code modified to accept 3.0.0 instead of 3.0.1.)

OBSERVATION:

When doing initial printer setup, if the Safety Timer fires, the printer gets in a weird state.

TO REPRODUCE:

Start initial printer setup.

Follow along, but get to a state where the printer is hot, but the printer is waiting for input.

Do nothing for a long time. Wait until Safety Timer fires.

Screen will say "Heating disabled by safety timer." with the choices being "Yes" and "No". Neither of those options make sense.

[image]

gudnimg, 2023-09-17:
Hi 👋 Can you elaborate what you mean by `initial printer setup`? The image you posted looks like the MMU Error screen is trying to render on top of the Safety Timer screen. (The clock symbol in far right corner suggests this to me)

trevorbattle, 2023-09-17:
Sorry. I was going through the procedure that starts with "I am your original Prusa i3 printer".

The printer is older and well proven, so the procedure should have been straightforward. I just walked away while it was happening and got distracted and didn't return for a long time.

Disabling heating was the right action, just the UI makes no sense.

closed 2023-12-11 by github-actions[bot]

Iv4nS, 2024-05-01:
this is still occurring, indeed the options of "yes" and "no" do not make sense and I don't know what I'm really answering to