prusa3d/Prusa-Firmware #4380: FW 3.13.1 filament detection not working at print start

jwingenroth, 2023-09-08:
**Printer type**  MK3S+
**Printer firmware version** -  3.13.1

**Describe the bug**
I normally send a print request from PrusaDSlicer without filament loaded. After mesh bed leveling, the print head goes to the front left to print the lead in line and detects there is no filament. Then starts the load filament routine.

After updating to FW 3.13.1 the printer starts printing with no filament loaded.

I confirmed the filament sensor is not indicating 1. All filament sensor settings are ON.

Down graded the FW and printed same file again and lack of filament is recognized.

gudnimg, 2023-09-12:
Could you give 3.13.2 RC1 a try? https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.13.2-RC1

We fixed a problem caused by an extrusion move in G80. It could cause false filament runout events for some users.

jwingenroth, 2023-09-23:
Results are the same with FW 3.13.2 RC1.
The printer did not detect a lack of filament at the start of the print.
Jerry

An4chr0n, 2023-10-13:
Can confirm the same behaviour with a Prusa mk3s+, firmware 3.13.1 just installed.
Filament detection not working and print starting without asking for it.

slackista, 2023-10-31:
Same with 3.13.2

tippmam, 2023-11-06:
Hallo, I was about to report this issue with my Stock MK3S+ on FW Version 3.13.1-6876 until I found this post.

> Menu > Settings > HW settings > Fsensor Detection Test does pass
> Menu > Settings > HW setup > FS action is set to Pause
> Support > Sensor Info > loaded and unloaded Filament is detected
> Cutting the Filament during a Print does pause print.

So basically everything is working like it should, excepting for the detection of filament on print start.

tippmam, 2023-11-07:
Update : Installed FW 3.13.2-7080 today. Problem remains.

jwingenroth, 2023-11-12:
I sent an email to info@prusa3d.com. Per the help.prusa3d.com this is how to get help with issues not covered within the website. I've pasted in the conversation below for the benefit of those with this issue and those who might stumble across it while evaluating Prusa support pre-purchase.

From this exchange several things can be deduced.
- Prusa Research either does not test the functionality of firmware changes before release or the testing done is inadequate.
- Prusa's responses to issues are treated with a shotgun approach. In this case suggesting multiple additional firmware updates before requesting fundamental troubleshooting steps.
- Prusa does not respond to simple and direct questions from their customers. I wonder if it is a real person or simply a bot responding in the email chain. There really is no difference, if it is in fact an actual person, if they simply follow a script and refuse to acknowledge the customer's inquiry.
- In my first email I provide the info for this bug report on Github. At that point several other posts stated having the same issue. I can't understand how it is that at that point Prusa would not confirm the issue on their end with an attempt to duplicate the issue with their hardware. Surely Prusa must have an in house means of testing customer complaints. This issue is easily duplicated and continued requests for action from me indicates my time is less valuable to them than the few short minutes they would spend confirming the issue.

I would like to take advantage of some of the recent changes in firmware. My workflow of loading filament after mesh bed leveling is the only technique I found to combat the oozing ( and resultant filament on the bed and/or nozzle) during mesh bed leveling.

I bought Prusa expecting a higher level of support. Clearly my expectations were unfounded. 

Over the last 18 months I've gotten to know this printer quite well and can work within it's limitations to produce reliable prints. I'll be looking quite seriously at other manufacturers when it comes time to buy my next printer.

I have submitted the requested videos today. Perhaps this will prompt some action toward a solution from Prusa. Not holding my breath.

Begin support email chain, reverse chronological order:
Nov 7.
Hello Jerry. 

To confirm this, can you please send me the picture of the printer's settings with the filament sensor being turned on? Furthermore, can you please record on a video with the look on the printer's sensor info that the value of IR sensor changes once you insert the filament? 

I greatly appreciate your help. 

Kind regards, 

—
Zuzana Ungerová
Customer support
PRUSA Research  |  Partyzánská 188/7a | 17000  Prague  | 00420 222 263 718 | 00421 220 570 305 | 00420 226 258 861
—
Your feedback is important to us!
Excellent!, 10/10Could be better, 5/10Awful, 1/10 
—
[Our 24/7 LiveChat](https://shop.prusa3d.com/en/) help is available in our eshop at the right bottom corner. 

Troubleshooting tutorials [here](https://help.prusa3d.com/)! Follow us on Twitter and [Facebook](https://www.facebook.com/PrusaResearch/)! Check out our [Blog](https://www.prusaprinters.org/)! Have a look at our [Service guides](https://manual.prusa3d.com/c/Service_manuals)!
Hide quoted text
Sent from Front
> On November 5, 2023 at 5:27 PM GMT+1 [jerry@wingenroth.us](mailto:jerry@wingenroth.us) wrote:
> 
> I have updated the firmware to 3.13.2.
> The printer does not detect the absence of filament at the start of the print and prints without filament.
> Yes, the issue persists.
> 
> Please help me understand something.
> 
> Is this issue repeatable by Prusa? There is nothing unique about my hardware. It has not been modified. I do not understand why Prusa doesn't replicate the issue on their hardware, develop a fix, and deploy it knowing they have tested and resolved the issue. It seems to me Prusa is guessing at a fix and they have guessed wrong 3 times now.
> 
> I bought Prusa expecting more in the way of support. I'm contemplating my next 3D printer hardware purchase and the experience trying to get this issue resolved makes me think I should invest my time and money in another brand. 
> 
> Please answer my single question in this email.
> 
> Jerry Wingenroth
> 
> On Tue, Oct 31, 2023, at 05:05, Zuzana Ungerová wrote:
>> Hello Jerry. 
>> 
>> Please try [the latest FW 3.13.2](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.13.2), and let me know if the issue persists. 
>> 
>> I greatly appreciate your help. 
>> 
>> Kind regards,
>> 
>> —
>> Zuzana Ungerová
>> Customer support
>> PRUSA Research  |  Partyzánská 188/7a | 17000  Prague  | 00420 222 263 718 | 00421 220 570 305 | 00420 226 258 861
>> —
>> Your feedback is important to us!
>> Excellent!, 10/10Could be better, 5/10Awful, 1/10 
>> —
>> [Our 24/7 LiveChat](https://shop.prusa3d.com/en/) help is available in our eshop at the right bottom corner. 
>> 
>> Troubleshooting tutorials [here](https://help.prusa3d.com/)! Follow us on Twitter and [Facebook](https://www.facebook.com/PrusaResearch/)! Check out our [Blog](https://www.prusaprinters.org/)! Have a look at our [Service guides](https://manual.prusa3d.com/c/Service_manuals)!
>> Sent from Front
>>> On October 26, 2023 at 12:51 PM GMT+2 [jerry@wingenroth.us](mailto:jerry@wingenroth.us) wrote:
>>> 
>>> 
>>> I've been using my MK3S+ for over a year now.
[... cut ...] t starts, mesh bed leveling is completed, and then the printer detects the lack of filament and starts the load filament routine before starting the print.
>>> This no longer works since FW version 3.13.0. The lack of filament is not detected and the printer proceeds to print without filament.
>>> 
>>> I submitted an issue, #4380, back on Sep 08. Others have commented and are confirming the issue.
>>> 
>>> Is Prusa Research going to address this change in functionality?
>>> 
>>> Jerry Wingenroth

An4chr0n, 2023-11-13:
@jwingenroth actually if you are using MK3S/+ you can avoid ozing with a Start G-Code (taken from Prusa discord help channel pinned messages): https://pastebin.com/5ZsWBBDK
You have to put the code into Prusa Slicer > Printer settings > Custom G-Code in the "Start G-code" text area.

twbaumann, 2023-11-22:
Same here with my MK3S+. I reluctantly updated from FW 3.12.2 to 3.13.2, hoping the issue was fixed or does not affect every printer. Now also my printer starts printing without filament loaded. I will downgrade to 3.12.2 until there is a fix available.

bjanders, 2024-02-04:
I'm not sure if it's related, but my MK3S did not detect it was out of filament when resuming from a pause, as I reported in #4482. I paused the print, pulled out the filament, and resumed printing expecting it to ask me to load filament, but it started printing without filament.

mobil750, 2024-02-06:
Today I experienced the same issue. The filament sensor seems to be completely ignored since FW V3.13.x.
Hope that Prusa reacts soon on this ugly issue ....

metteo, 2024-02-11:
I can confirm the problem. Had to downgrade all the way to 3.11.0 because of another issue (lcd glitch) and it also fixed the problem with unload/load during print start. Considering the fact that MK3S+ will not be produced after Feb 2024 it would be nice get some fixes.

Prusa-Support, 2024-02-27:
Thanks for your feedback.
This issue has the developers' attention.

Michele Moramarco
Prusa Researchh

Ricky0351, 2024-03-16:
issue still there with 3.13.3

alessandroceleste, 2024-05-12:
Thanks for reporting the issue. I realized the issue yesterday with 3.13.3-7094 and wanted to report, too. Waiting for a fix ...

Prusa-Support, 2024-05-25:
> I only load the filament after mesh bed leveling as a workaround to the oozing...

Keeping the filament very dry should make a noticeable difference.
Other workarounds may be:
withdrawing the filament by a couple of millimeters after loading;
letting it ooze slightly longer before cleaning and continuing with the print;
in some cases you may need to set a very slightly lower first-layer nozzle temperature;
not to mention tweaking the start g-code to slightly modify the temperatures before/after MBL is possible too.

.

The new (current) behavior is intended to reduce the chances of fake filament triggering in specific scenarios but our developers are considering alternative solutions to fix the new issue.
They will probably find the resources to do that after the release of FW 3.14.0.
For now, this issue is set as a milestone for FW 3.14.1.

Michele Moramarco
Prusa Research

jwingenroth, 2024-05-25:
"to reduce the chances of fake filament triggering in specific scenarios" What the heck does that mean?

I assumed this was an unintentional bug induced by the developers.

Are you saying this was an intentional, unannounced , filament detection behavior change?

I don't need more workarounds. I have a work flow that works quite well with FW 3.12 . Who knows what other undocumented changes have been  made in the versions since.

I understand that tweaking settings is a part of getting good prints but now realize that successful workflows could be rendered no good with any future firmware update.

Disappointing Prusa.

slackista, 2024-05-26:
LOL. Filament detection not working at print start. Prusa support 24/365/∞: 
>Keeping the filament very dry should make a noticeable difference.
@jwingenroth which part you don't understand?

hkramer70, 2024-05-27:
It is very sad that such a basic detection routine (at the beginning of printing or while printing), which, BTW, is part of Prusas documented workflow, was simply removed and nothing happens since last year september/october to reactivate that required functionality...

Mark-FS, 2024-06-01:
Can I ask if you guys have the new IR filament sensor (3 wires) or the original one (4 wires).  Mine is the original sensor and is completely dead on all FW after 3.9.3.  I therefore have a choice of a working filament sensor with 3.93 or installing 3.30.1 and having Prusa Link working.

I have been advised that my problem is the same as this thread but that would mean the filament sensor has been broken for almost 3 years (3.10.1 was released in 2021) and that doesn't sound right.  Thanks.

Prusa-Support, 2024-07-23:
> What the heck does that mean?

I know... The "heck" is intended to avoid other problems, and I'll try to clarify based on what I learned but I'm afraid I won't be able to provide a complete piece of information - FYI @leptun.

I'll try to explain what I know, trying to be as concise as I can be, but I think this is only a partial explanation and I apologize in advantage if this may once again sound misleading. In case of doubts, or if you have other opinions to share, a polite comment would be all.

**Old behavior**: the filament sensor was being checked all the time and filament outage could be accidentally triggered by certain unintentional movements. As an example, it could happen during user manipulations.
**New behavior**: the filament detection is only triggered in the event of transitioning _from `filament detected` status to `filament undetected` status_ to prevent accidental and potentially troublesome triggering. It is a bit of an edge case so if haven't come across it, I'm glad you haven't experienced this : ) and with the current behavior the chances of accidental triggering are far lower.

> I assumed this was an unintentional bug induced by the developers.

All in all, this statement is not accurate but what I was trying to say is that **our developers are more than willing to improve the current behavior** (or make it optional).

> Who knows what other undocumented changes have been made in the versions since.

As a matter of fact, probably more than what our developers can keep on the radar and fit in the release notes but the firmware is open source so everything is discoverable on GitHub. I'm not a developer myself and I barely understand the firmware release comparison tools but anyone on GitHub can take advantage of that, so you have my understanding and I hope this helps.

Finally, allow me to reiterate that this issue is set as a milestone for FW 3.14.1.
Our developers are on it.

Michele Moramarco
Prusa Research

jwingenroth, 2024-07-23:
Thank you for attempting to explain.

closed 2024-08-30 by 3d-gussner

metteo, 2024-08-30:
@gudnimg @3d-gussner I'm not sure if #4727 solves the issue mentioned here (in the issue description / comments). 

One of the important features of the previous implementation was the ability to start the print **without** the filament, pass the calibration (mesh bed leveling) and then go through unload / **load procedure** (unload was unneeded but still, the load happens afterwards). This is really helpful because prevents bits of filament ending up on the bed during the calibration. 

With the new implementation you have to **load the filament** and then **start the print** (meaning I have to take some time cleaning the nozzle from oozing filament after load before I can start actual print)

@jwingenroth could you confirm as you are the author of the issue?

jwingenroth, 2024-08-30:
@metteo when I read the description of #4727, it does NOT seem to address the issue I raised in #4380.

"This is really helpful because prevents bits of filament ending up on the bed during the calibration." This is exactly the issue I struggled with. I am sure that I found a response from Prusa in the comments section on the prusa3d web site suggesting the sequence of unloading the filament after each print and loading after mesh bed leveling. I've looked but I can't find it again.

I've pretty much given up on Prusa. The MK3S+ was my first printer and I learned a lot using it. I'm able to get decent prints with my 2 MK3S+ printers stuck at lower firmware levels. I've moved on with another manufacturer and am very please with the results I'm getting with them. It seems like a waste of my limited time to try and get Prusa to come around on this issue.

hkramer70, 2024-08-30:
I also raised the same issue as a ticket here when they changed that firmware. But I also had not success/got no satisfied response. I gave up. Disappointed by the manufacturer.

Prusa-Support, 2024-09-24:
I'm sorry for the confusion but I think the issue is actually solved. I hope you may give it a try and report back when FW 3.14.1 is released. It seems to be close to be released actually so please give it a look at https://github.com/prusa3d/Prusa-Firmware/releases in a few weeks, or days!

Michele Moramarco
Prusa Research