prusa3d/Prusa-Firmware #4384: [BUG]Tune-Temperature not display in German Language>

Drake0815, 2023-09-12:
**Printer type** - [MK3S+]
**Printer firmware version** -  [e.g. 3.13.0, 3.13.1]

**SD card or USB/Octoprint**
  Print via SD card

**Describe the bug**
  With the Upgrade from Firmware 3.12.2 to 3.13.0 and also in Version 3.13.2 is a Bug related to the Tune Menu when we use German Language.
  When we change the Printer Language to German in the Tune Menu the Entry Speed is not showing right. when you turn the dial wheel it flashes briefly
  
**To Reproduce**
 Set the Language to German Language start a Print from SD Card and go to the Feineinstellung Menu during this.
  
**Video**
  Please attach a video. It usually helps to solve the problem.
[image]
[image]

sarusani, 2023-09-12:
Did you update the firmware via PrusaSlicer or via Octoprint?

Octoprint can't update the language files for the translations and this string was updated lately...

Drake0815, 2023-09-12:
@sarusani via PrusaSlicer

sarusani, 2023-09-12:
Ah, sorry. My mistake.

The error has already been fixed, but was not released in 3.13.1. It's fixed in [3.13.2-RC1](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.13.2-RC1) which was just released.

3d-gussner, 2023-09-13:
@Drake0815 Please consider to close this issue as it has been solved. Bitte schliesse das Ticket da wir dies in der FW 3.13.2-RC1 behoben haben.