prusa3d/Prusa-Firmware #4392: [BUG][MK3S+][MMU2S] FINDA triggering issues since FW 3.0.x

ptashek, 2023-09-16:
Possibly related to issue [4351](https://github.com/prusa3d/Prusa-Firmware/issues/4351)

**Printer type** MK3S+
**Printer firmware version** -  3.13.2-RC1

**MMU upgrade** - MMU2S
**MMU upgrade firmware version** - 3.0.1-RC1

**Slicer version**
PrusaSlicer 2.6.1+win64

**SD card or USB/Octoprint**
SD card

**Describe the bug**
  After upgrading MMU FW from 1.0.6 to 3.0.x the MMU is consistently running into filament loading/unloading issues, related to the FINDA sensor not triggering. The two most common errors are: FINDA Filament Stuck (err 04102), and FINDA did not trigger. Having checked, the filament is never actually stuck, and always read to load (i.e. parked correctly). There are no obstructions in the MMU, the feeder tubes are all clean and filament can move freely.
  
 It could be a failing FINDA, but it's a coincidence that the issue started right after moving to FW 3.0.0.
  
**To Reproduce**
I get both errors during the loading/unloading stage of every MMU print, but only at the start or end of the print. During a print filament unloading / loading does not seem to trigger this issue, and tool changes happen without intervention.
  
**Expected behavior**
At the start / end of a print filament is loaded / unloaded without errors.
  
**G-code**
In my case, these errors reliably trigger for any MMU gcode.

gudnimg, 2023-09-16:
Hi 👋 The new FINDA error codes are very reliable so far in my experience and I think it's most likely the sensor is not positioned/calibrated optimally (could also be a problem with pulley gears)

FINDA Filament Stuck (`FINDA_DIDNT_SWITCH_OFF`) happens on unload usually and it means the pulley gears could not pull the filament back such that the FINDA stopped detecting filament. Can be many reasons for this => 1) a lot of friction in the PTFE tube path 2) filament is grinded by pulley gears 3) Idler/pulley are not gripping the filament enough (note this could also be debris in the pulley gears) 4) **FINDA is positioned too high (most likely IMO)**

FINDA did not trigger (`FINDA_DIDNT_SWITCH_ON`), is usually always the FINDA sensor needs to be lowered slightly. I've seen issues on my MMU2S unit where if the filament is bent sharply (like at the beginning of a print Z=0mm or at max Z) then FINDA would not trigger correctly.

gudnimg, 2023-09-16:
You could also try using the Loading Test menu which is located in `Settings -> Load Test` this will attempt to load filament into the extruder and then also performs an unload.

gudnimg, 2023-09-16:
Also another note. On the MMU error screen on the LCD, the `FI: 0` and `FI: 1` show what your FINDA sensor is measuring. 

If you see `FI: 1` when you get FINDA Filament Stuck, then the firmware is correctly reporting the issue. Same applies if you see `FI: 0` when you get FINDA did not trigger, then that indicates your FINDA needs positioned/calibrated better.

You can also see on the serial logs from the printer the status of the filament sensor readouts.

ptashek, 2023-09-16:
Thank you for the extensive details. I have managed to resolve the issue, however I still think this may be somewhat related to the new MMU FW. These are the steps I have followed:

- detached the MMU <-> hotend tube
- checked FINDA values in  _Support -> Sensor info_ while moving a piece of filament past the sensor
- FINDA was triggering, but every now and then would not and I have also noticed there's quite a delay (>1s)
- I've loosened the FINDA mount, and moved the sensor down at first - same behaviour
- Then I've moved the sensor up, and noticed that at some point it started reacting considerably quicker and every time
- After fixing it in this position, I cannot reproduce the load/unload issue anymore

What's interesting here:
- the FINDA hasn't been adjusted since I've put the MMU2 together, and it couldn't have moved by itself
- this issue was not present with FW 1.0.6, but started immediately after installing FW 3.0.0 and persisted with FW 3.0.1

Given the issue is resolved for me following your advice and the manual, I'll leave it up to you to decide whether any further investigation is warranted given this is new (for me) behaviour in FW 3.x. Feel free to close the issue, if not :)

gudnimg, 2023-09-17:
Closing the issue as solved for now.

closed 2023-09-17 by gudnimg

ptashek, 2023-09-17:
Looks like my optimism was premature. All worked fine for two smaller MMU prints. Today I've ran a MMU-single print (gCode attached), and the unit behaves in the most random way I have yet seen.

About 3/4 into the print the MMU decided that it needs to unload F1 (used for the print), and load F2. Despite F1 unloading correctly and FINDA sensor triggering correctly (checked, and I'm 100% sure), the MMU claims the sensor did not trigger, THEN despite that, it moves to F2 and tries to load it. There's no filament to load, of course. As a result the FINDA 04101 error triggers again. At this point, there is now way to resume the print at all. The MMU is stuck in a loop trying to load F2.

[Box_0.6n_0.3mm_PETG_MK3SMMU2S_3h6m.zip](https://github.com/prusa3d/Prusa-Firmware/files/12643165/Box_0.6n_0.3mm_PETG_MK3SMMU2S_3h6m.zip)

gudnimg, 2023-09-17:
Do you have SpoolJoin on? This behavior is expected if the printer encounter filament runout on the FINDA sensor. 

With SpoolJoin off you would instead be prompted to select the next slot to load. While SpoolJoin always selects the next adjacent slot.