prusa3d/Prusa-Firmware #4393: [BUG]<Enter comprehensive title>

bryn51, 2023-09-17:
> Check <

**Printer type** -Mk3S+ with MMU3
**Printer firmware version** -  3.13.0 -6873

**MMU upgrade** - MMU3
**MMU upgrade firmware version** - 3.0.0

**SD card or USB/Octoprint**
  PrusaLink (RPi Zero ...)

**Describe the bug**
  When using the printer LCD Screen, I see sometimes that the screen displays garbage (non alphabet characters) filling the whole screen. It appears at (apparently) random times, I have not yet picked a pattern. It goes away after a short time, and the display shows normally.  This was happening before I installed the RPI Zero.
  
**To Reproduce**
  I have not yet picked a pattern
  
**Expected behavior**
  I expected the screen to display the firmware menu at the current selected level.
  
**G-code**
This does not require gcode running to occur.

**Video**
  This happens fleetingly, at unpredictable times.  I have not yet had the chance to capture it on video.

sarusani, 2023-09-17:
Check the ribbon cable that connects the display to the board for defects and make sure the connection on both sides is proper. I'd guess this is likely a loose connection. (Probably on the LCD side, because it happens when you use the nob)

It doesn't sound like a software bug to me. I'd ask [customer  support](https://help.prusa3d.com/article/customer-support_2287) for assistance instead.

bryn51, 2023-09-17:
Erm, no it does not happen only when i use the knob. It happens when using the knob, and when I have my back turned as well.And I already checked the cables before making my report. Please stop making rash assumptions, and politely  ask me instead.Sent from my iPhone

sarusani, 2023-09-17:
Where did you mention that you already made sure the cable and the connections are fine? How should we know this?
I just wrote what in MY OPINION is the most likely cause. That's all I can do to try to help you out.

But still, this doesn't sound like a software bug to me. (And I could totally be wrong about that! But since nobody else reported a similar issue, it's not unreasonable to believe that.)

Customer support has much more experience in evaluating problems with the printer. Maybe it's a problem with the board itself, maybe a problem with the LCD module. The people to know best are the ones from customer support.

bryn51, 2023-09-17:
Its incorrect to say other people have not reported this issue. Maybe on this system I dunno, since I searched for it  and found no other reports. However this issue has been happening to me since installing 3.13.0 and not before. I chased the wiring thru to the Einsy Rambo and found nothing out of order.I thought it might be a intermittent issue, which might mean something on the way out.  So, I kept it to myself until, on Facebook, a couple of other folks reported the same thing.  i kept it to myself because the issue quickly went away, then came back later (and again rinse repeat).Until those other folks reported the same thing.I grant that the firmware is tested extensively prior to release, but that does not negate the possibility of bugs remaining. You have to know that I am a trained electronics technician and also systems analyst/programmer, for over 30 years.If you are not from Prusa Research, I do not need your help, thank you. CheersBrynSent from my iPhone

sarusani, 2023-09-17:
> Its incorrect to say other people have not reported this issue. Maybe on this system I dunno...
That's exactly why you should contact customer support. They know if this has been reported before and if/how it was solved.

>If you are not from Prusa Research, I do not need your help, thank you
Another reason to contact customer support.
This is a repo for the open source firmware, it's maintained by Prusa and the community. If you only want to speak to Prusa employees, they are reachable 24/4 in the [live chat](https://www.prusa3d.com/). (Look for the "Chat now" on the Bottom right.)

gudnimg, 2023-09-17:
I agree this issue should be raised with @Prusa-Support unless we have some steps to reproduce it.

cqeta1564, 2023-09-20:
I have the same issue. It hapends after any unpredicted reset (panic button, short termistor or heater “thing”, …). Only way to remove it is to wait. Sometimes for 5 minutes and sometimes for 40 minutes!!! I don’t know why it’s happening, but it started after updating to 3.13.x. Same hardware as you.

cqeta1564, 2023-09-20:
Adding picture.
[image]

cqeta1564, 2023-09-20:
Uploading picture. 
[image]

gudnimg, 2023-10-28:
@bryn51 @cqeta1564 Hi 👋 , I wanted to check in and ask if you are still seeing this issue?

cqeta1564, 2023-10-28:
Hi, no I am not. Sorry that I forgot to write that here. I changed the thermometer and heater block and downgraded the software. Now it works without any problems.

bryn51, 2023-10-28:
> @bryn51 @cqeta1564 Hi 👋 , I wanted to check in and ask if you are still seeing this issue? 

Hi

Well I had not seen this happen in a while, but a day or two ago (sometime after recent upgrade of firmware to 3.13.1) I saw the garbled screen again. 

In recent time I have been using PrusaLink to send gcodes to the printer therefore less time spent sitting in front of the printer. Thus less likely to observe this issue when/if it occurs.
Also as time has elapsed the printer has been in and out of enclosure a number of times for servicing, so screen has been disconnected/reconnected. Now, the ribbon cable && connectors are a prime candidate if this issue is hardware as claimed here. So you'd think perhaps the nature of this issue might have changed by that means, but no, it remains. I have also checked the einsy end of connection.

bryn51, 2023-12-28:
Well it's true that I have not seen this issue in quite some time.

But my interest has waned somewhat as of today since I received notification that my mk3S+ > mk4 upgrade kit for this printer has been shipped. So the motherboard, wiring, display and firmware will all be changed out. The old items will be sold off as spares to someone else, or possibly up-cycled to make a filament recycler.
It appears like only one person looked at this report seriously enough to try to reproduce it.
My thoughts are that Prusa Research need to reconsider how bug reports are handled. Either Github reporting is supported or it's not. If it's not then it ought to be shut down as it's counter productive. If it is supported then it needs to be run better and with greater transparency. The reports should be maintained so that persons reporting bugs and making enhancement requests know where it's at.  Prusa Support can/should be involved as a second channel of support. Where an issue requires further evidence they could/should reach out to customers.
The kind of gruff responses seen on this thread  should never occur, as they tend to discourage folks from reporting issues, an undesirable outcome. 
Customers are the final testing platform for firmwares and cohtribution ought to be welcomed and encouraged. The Prusa Support chatline channel is surely unsuitable for handling matters like this. Granted, PS staff can act as triage for issues, but ultimately after reporting issues customers want to know about progress toward addressing issues, and that feedback is sadly lacking, no matter what channel is used for reporting.

Prusa-Support, 2024-02-27:
This is usually caused by external signal interference, exposed wires, or loose connectors - among other rather elusive causes.

The MK3 printers are provided with basic Electrostatic discharge (ESD) protection but such protection is limited to a certain level of charge and hardware/ambient conditions. 

Again sometimes the reason for electrostatic interference can be found in the wiring but in certain conditions, even a human body may be significantly charged from the environment tension so that when touching the printer the body will discharge the tension through the machine.
_It may not be a coincidence if the LCD panel is subject to this kind of erratic glitches as it is the main interface between printer and human._
These charges are harmless but if the ambient conditions are specifically favorable for this phenomenon, touching the PSU before touching any other part of the printer may help.

It goes without saying that the correct grounding of the printer is important and would reduce the chances of interference within the electronics.

This can't be improved via firmware. However, with the new series of printers, we are introducing an improved system of Protective Earthing (PE).
This said, ultimately I'm afraid we are not looking at a firmware bug so I would recommend closing the issue.

Michele Moramarco
Prusa Research

bryn51, 2024-02-27:
You are likely correct. Not firmware but hardware. Correct just to close the issue ?!
Noooo. Whilst the issue may have been addressed in some respect in the mk4, which at least has ground wiring, it remains in the mk3 which does not. All solved by obsolescence of the mk3 right ?
But awfully strange that I found it occurred more often in new zealand country town (with relative radio quiet) than in Taiwan where I lived in a large city not far from a military airfield. The induced electrical noise in TW was measured at 150 mV but this issue did not occur even once.
- Lets hope the issue no longer appears on the mk4 I have in place of the mk3 I had when this issue was reported.

bryn51, 2024-05-13:
It seems to me that this issue likely has its root cause in fluctuating random electrical noise, and it may have been addressed by Prusa in the Mk4 printer which has a proper earth connection to the LCD,.  But not addressed in the MK3S at all, and I guess this printer is regarded as obsolete now.
In any case it is not interesting to me any more, as I upgraded mk MK3S to a mk4.  Of course it may interest many others who have retained the MK3S in their stable.

3d-gussner, 2024-07-08:
@bryn51 I will close this issue, if you still think it is relevant please feel free to reopen it.

closed 2024-07-08 by 3d-gussner