prusa3d/Prusa-Firmware #4395: [BUG]PrusaLink doesn't start with MMU 3 activated

calibx, 2023-09-18:
**Printer type**
MK3S+
**Printer firmware version**
3.13.1

**MMU upgrade**
MMU 3
**MMU upgrade firmware version**
3.0.0

**SD card or USB/Octoprint**
Both (startup problem)

**Describe the bug**
When powering on the printer the display stay on "PrusaLink starting" and never continue.
If you go to the settings and disable MMU, PrusaLink start and you can re-enable MMU and use it with PrusaConnect
On the PrusaLink site when the MMU is enable at startup there is an connection error to the printer.
  
**To Reproduce**
Power on the printer with MMU and PrusaLink on PI Zero
  
**Expected behavior**
Printer and PrusaLink start with MMU activated
  
**G-code**
None

**Video**
Coming soon

TojikCZ, 2023-09-18:
Hi, yes, pleas flash the PrusaLink 0.7.1RC or wait for the update. In the meantime, disable the mmu in FW, let PrusaLink connect and enable it again after. Sorry, the release is held up by pip package name conflict of a sub-library

calibx, 2023-09-18:
Fixed with the 0.7.1RC !

closed 2023-09-18 by calibx