prusa3d/Prusa-Firmware #4400: [BUG] Printer displays unrelated and misleading error messages when Power Panic is the issue

aeberbach, 2023-09-21:
MK3S - all firmware versions

When Power Panic condition is detected printer does not display any meaningful indication that this is the cause of self-test failure.

Following replacing a power supply I connected the original black PSU to the MK3S connecting the power panic lead. Printer then failed self test at the heatbed stage - heatbed light did not illuminate, heatbed did not heat.

Attempts to manually set heatbed temperature to check operation result in "THERMAL ANOMALY".

Printer should display some meaningful indicator that power panic has triggered! Failure of heatbed or nozzle heating or THERMAL ANOMALY gives no indication of the actual issue and makes finding and fixing it unnecessarily difficult!
  
**To Reproduce**
Trigger power panic, run self-test or set heatbed or nozzle temperature manually.
  
**Expected behavior**
Error message indicating POWER PANIC and not THERMAL ANOMALY or other unrelated message.

aeberbach, 2023-09-21:
Surely self-test would be one time that the user should be alerted of unexpected power panic status?

gudnimg, 2023-09-22:
There are a lot of scenarios where power panic feature does not really do much. You raise an interesting point => if a power outage happens while the printer is running self-test OR any calibration, then it would be useful to dismiss any saved results and 'invalid'.

Currently power panic feature is ONLY used when printing and is used to recover the running print. But this doesn't mean we can't expand it.

aeberbach, 2023-09-24:
Is it correct to say that nAC_FAULT is triggered when power panic is detected - even though the printer is trying to turn the heatbed on, the AND gate keeps it turned off?

gudnimg, 2023-09-24:
Yes `nAC_FAULT` is the signal which triggers the power panic hardware interrupt when a falling edge is seen.

This function will log `"INT4"` via UART

```c
ISR(INT4_vect) {
    EIMSK &= ~(1 << 4); //disable INT4 interrupt to make sure that this code will be executed just once
    SERIAL_ECHOLNRPGM(MSG_INT4);
    if (eeprom_read_byte((uint8_t*)EEPROM_UVLO) == NO_PENDING_RECOVERY)
    {
        if(printer_active()) {
            uvlo_();
        }
    } else {
        // There is already a pending recovery waiting. Power outage in this scenario
        // arrives before we can fully recover the print. In that case call a 'tiny'
        // version of uvlo_() which doesn't overwrite the print state already waiting in EEPROM
        uvlo_tiny();
    }
}
```

> even though the printer is trying to turn the heatbed on, the AND gate keeps it turned off?

Yes, from the schematic below it looks like to me `nAC_Fault` signal turns off the heaters when a power panic is detected.

https://github.com/ultimachine/Einsy-Rambo/blob/1.0a/board/Project%20Outputs/Schematic%20Prints_Einsy%20Rambo_1.0a.PDF
[image]
[image]
[image]