prusa3d/Prusa-Firmware #4401: Firmware 3.1.3.2-RC1 & 3.0.0: MMU FW update needed at MMU start up > Reset or Disable MMU message

mmoraes80, 2023-09-23:
**Printer type** - [MK3S]
**Printer firmware version** -  [3.13.2 RC1]

**MMU upgrade** - [MMU2S]
**MMU upgrade firmware version** - [3.0.0]

**SD card or USB/Octoprint**
  PrusaLink 0.7.0

**Describe the bug**
  After flashing MK3S firmware 3.13.2-RC1 and enabling the MMU under Settings menu, a message in the LCD display is shown stating that the MMU needs an update. Only options are Reset MMU or Disable it. Printer works just fine with MMU turned OFF.
  
**To Reproduce**
  Flash FW 3.13.2-RC1, turn MMU on. Message will appear after MMU boots up. Also happens when the printer is powered off and back on with MMU already enabled.
  LCD Message:
    MMU FW UPDATE NEEDED 
    prusa.io/04504
    FI: 0 FS: 0 F? Temp 23
    ResetMMU >Disable
  
**Expected behavior**
  Normal start up procedure.
  
**G-code**
  N/A.

**Video**
  N/A.

sarusani, 2023-09-23:
Firmware 3.13.2-RC1 requires MMU firmware [MMU3_3.0.1](https://github.com/prusa3d/Prusa-Firmware/releases/download/v3.13.2-RC1/MMU3_3.0.1+837.hex).

gudnimg, 2023-09-23:
Closing since this is not an issue

closed 2023-09-23 by gudnimg

gudnimg, 2023-09-23:
Just a note: If you're not sure if a new version is required. When you click the double down arrow symbol ⏬ on the MMU error screen to see more description for the error) you should see a text which shows exactly which MMU FW version is needed.

We force users to update since some functionality on the printer side may expect some changes on the MMU FW side as well.