prusa3d/Prusa-Firmware #4406: [BUG] Language "Deutsch" not working in 3.12.x

C-H-F, 2023-09-23:
**Printer type** - MK3S+
**Printer firmware version** - 3.13.0, 3.13.1

**SD card or USB/Octoprint**
 USB/Octoprint

**Describe the bug**
  After installing the firmware 3.13.x the printer asks for the language "English, Cestia, Deutsch, Espanol". After selecting "Deutsch", the printer shows the message "Copy selected language?". Choosing "Yes" now opens the language prompt again.
  Choosing "English" works without problems.
  
**To Reproduce**
  - Upgrade from 3.12.2 to 3.13.x.
  - Select "Deutsch"
  - Choose "Yes"
  - Now you have to choose your language again.  

**Expected behavior**
  After choosing "Deutsch" and "Yes" the message "Original Prusa i3" should appear and the printer should start to work.

gudnimg, 2023-09-23:
Hi 👋 please try 3.13.2 RC1. https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.13.2-RC1

What you describe sounds a lot like one bug we fixed.

C-H-F, 2023-09-23:
Tested it with the built version prusa3d_fw_MK3S_MK3S+_3_13_2-RC1_7068.hex.
Now there is a new step after choosing "Yes". The printer shows  "Language update.. Original Prusa i3"... but it still goes back to the language selection.

3d-gussner, 2023-09-24:
@C-H-F Did you flash the printer with **Prusa Slicer**?

In case you used the Octoprint firmware updater **DON'T** use it as that one doesn't update the in xflash stored translations on MK3/S/+.
There is a PR https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/pull/325 for that but not implemented yet.

C-H-F, 2023-09-24:
Thank you very much.
It's really a problem of *OctoPrint* and not the firmware itself. It works perfectly when flashing through Prusa Slicer.

closed 2023-09-24 by C-H-F

3d-gussner, 2023-09-24:
@C-H-F Glad to hear that it isn't the Firmware. Thanks for the feedback and closing the issue, many users don't close it even it has been solved. Maybe you can ask the octoprint firmware update developers to consider to use the PR which should solve the issue.

Viel Spaß beim drucken.

tm8, 2023-11-22:
Same here on Version 3.13.3_7080.

[https://github.com/prusa3d/Prusa-Firmware/issues/4406#issuecomment-1732506641](https://github.com/prusa3d/Prusa-Firmware/issues/4406#issuecomment-1732506641)