prusa3d/Prusa-Firmware #4412: "Set Ready" Prusa Connect button in menu of MK3

huskyte, 2023-09-27:
A button to pass "set ready" to PrusaConnect in order to continue on the print queue would be needed on the menu of the physical printer if the printer and the PC are separated by each other. In my case the printers are in another room. So, if I wated to see if everthing was alright I would clean up the printer, run back to the PC, set the printer ready in connect, back to the printer, have a look and be back again. 
A simple "set ready" in the info screen where eg "unload filament"resides would perfectly do.

**Printer type** - [e.g. MK3S, MK3, MK2.5S, MK2.5, MK2S, MK2]
**MMU Upgrade** - [e.g. MMU2S, MMU2, MMU1]

3d-gussner, 2023-09-27:
@huskyte Thanks for reporting it. We are working on that.
The firmware 3.14.0 is focused on to improve the PrusaLink - PrusaConnect with MK3/S/+ printers.

closed 2023-11-22 by 3d-gussner

darraghbr, 2023-12-01:
@huskyte 

There has been a method to do this since shortly after PrusaLink came around. Have a look at this forum post.

https://forum.prusa3d.com/forum/introcuctions-instrutctions/set-printer-ready-on-printer-itself/#post-686740

You just create a folder on the printer SD card named "PrusaLink menu", then put the SD back inside the printer and a file gets automatically generated called SETREADY.G that sets your printer to ready. 

I'm happy that there is a more formal way to do this coming but this method works just fine.