prusa3d/Prusa-Firmware #4413: Prusa MK3S+ Display anzeige

Dedalox, 2023-09-27:
Seit den letzten Update 3.13.1 wird bei mir beim Display die anzeige für die Geschwindigkeit nicht richtig dar gestellt.
Wenn ich wehrend des drucks mit den dreh Knopf durch das Menu scrolle steht bei mein Drucker MKS3+ unter den Feineinstellungen keine Geschwindigkeit in menu mit wieviel Prozent gerade der Drucker Druckt, es erscheint nur ein kurzes flackern dort wo die Geschwindigkeit angezeigt werden sollte dann.

:-). Dedalox

sarusani, 2023-09-27:
Der Fehler wurde in der Version 3.13.2 bereits behoben. Du kannst entweder den [Release Kandidat ](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.13.2-RC1)  ausprobieren oder auf die finale Version warten.

Dedalox, 2023-09-27:
Ok Danke für die Info!

closed 2023-09-27 by Dedalox