prusa3d/Prusa-Firmware #4414: [BUG] Too much MK3S<->MMU2 communication logging on serial port

thess, 2023-09-27:
**Printer type** - MK3S
**Printer firmware version** -  3.13.1+

**MMU upgrade** - MMU2S
**MMU upgrade firmware version** - 3.0.0, 3.0.1

**SD card or USB/Octoprint**
  Mainly USB/Octoprint

**Describe the bug**
 Serial communications between the printer and MMU are continually output on the USB port.
  
**To Reproduce**
  Connect to USB serial and notice continual chatter.
  
**Expected behavior**
 We need a way to suppress these messages from log output either by:
1. Providing menu option to disable (LogRequestMsg, LogResponse)
2. Conditional DEBUG build
3. Remove altogether
 
I suppose one could write an Octoprint serial filter to remove the transactions from the console but I would prefer not to have the excess data always clogging the serial communications path.

gudnimg, 2023-09-28:
I agree with this. It would be nice to hide the Query (Q0) if nothing has changed since the last Q0.

3d-gussner, 2023-09-28:
@thess Nice to see you back on this repo. We have been discussing this few times and we gonna keep it for the moment.
Reason to keep it is to be able to check in the serial logs what happens BEFORE an error occurs. We don't have the resources to keep some lines in the buffer to read this buffer when an error is reported.
If you have an idea how we could ensure that we get access to the communication between MMU3 and printer please let us know.

Here my OctoPrint Filter for the Terminal output:
Settings -> Terminal Filter -> Name: `Suppress MMU` RegExp : `echo:MMU2`

Some parts of my `config.yaml`
```
....
feature:
  autoUppercaseBlacklist:
  - M117
  - M118
  - M707
  - M708
  - M850
  - D3
  
...

terminalFilters:
- name: Suppress temperature messages
  regex: '(Send: (N\d+\s+)?M105)|(Recv:\s+(ok\s+([PBN]\d+\s+)*)?([BCLPR]|T\d*):-?\d+)'
- name: Suppress SD status messages
  regex: '(Send: (N\d+\s+)?M27)|(Recv: SD printing byte)|(Recv: Not SD printing)'
- name: Suppress position messages
  regex: (Send:\s+(N\d+\s+)?M114)|(Recv:\s+(ok\s+)?X:[+-]?([0-9]*[.])?[0-9]+\s+Y:[+-]?([0-9]*[.])?[0-9]+\s+Z:[+-]?([0-9]*[.])?[0-9]+\s+E\d*:[+-]?([0-9]*[.])?[0-9]+).*
- name: Suppress wait responses
  regex: 'Recv: wait'
- name: Suppress processing responses
  regex: 'Recv: (echo:\s*)?busy:\s*processing'
- name: Suppress MMU
  regex: echo:MMU2
```

gudnimg, 2023-10-19:
@thess I have a working pull request which shrinks the logging activity quite a bit. My proposal is to allow the full logging activity while the MMU is running a toolchange or any other command. But while it's sitting idle (e.g. after the toolchange is done) none of the "spammy" messages will get logged.

Would this be an acceptable solution in your opinion? My proposal only costs around ~40 bytes of flash memory.

thess, 2023-10-19:
I'll give your PR a try next week and see how it goes. I am using the Octoprint MMU2 plugin now in conjunction with serial filters. You proposal looks like the best compromise (so far). 

I also did a version of the firmware for my own use with the MMU serial logging disabled but then the Octoprint plugin is not usable.

thess, 2023-11-14:
Sorry for the late response -- Suppressing logging when idle is an acceptable solution. The Octoprint plugin is definitely usable. There may be some status info missed by the plugin at startup but I haven't done enough checking.

OK to close this if #4455 is merged.

closed 2023-12-28 by 3d-gussner