prusa3d/Prusa-Firmware #4417: Unable to upgrade MMU2S firmware to 3.13.*

tbatur, 2023-09-29:
I have 2 MK3S+ (with MMU2S) printers. Before version 3.13.*, when I updated the printer firmware, the MMU2S firmware was also updated automatically. 

But when I upgrade to this latest version (3.13.0, 3.13.1, 3.13.rc2) only the printer is updated, and when I activate the MMU in the settings it says that I have to update its firmware too.

I have tried this on both my printers with 5 different cables, I even bought 1 new data cable, and there is no problem with data exchange. I also tried it on both macOS and Windows but the result is always the same. It gives the following error. I have also contacted live support about this but they have not found a solution either.

As a result, I had to revert back to 3.12.2. I don't know if anyone else has this problem, but I couldn't upgrade.

[image]

henrikkoch, 2023-09-29:
You can just update the MMU2 with a cable directly in the MMU2 from
PrusaSlicer with the MMU2 hex file

tbatur, 2023-09-29:
> […](#)
> -- Med venlig hilsen / Kind regards *Henrik Johansen Koch*Hans Erik Nielsens Vej 7, DK 3650, Ølstykke, Denmark *T +45 4710 0305* *F +45 7734 0307* *M +45 4068 0305**E-mail ***@***.*** ***@***.***>*

I guess you didn't read exactly what I wrote. That's exactly how I tried it.

sarusani, 2023-09-29:
Looks like you try to update the MMU with the printer firmware. You need to download the [separate hex file](https://github.com/prusa3d/Prusa-Firmware/releases/download/v3.13.2-RC1/MMU3_3.0.1+837.hex) for the MMU. (The one I linked is compatible with [Printer Firmware 3.13.2-RC1](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.13.2-RC1))

henrikkoch, 2023-09-29:
Sorry. I just answered what you wrote: "Before version 3.13., when I
updated the printer firmware, the MMU2S firmware was also updated
automatically"

I have never seen this been done automatically. I have always done this
with a micro usb cable from the PC to the MMU2+.
And with the last version, I also did a "manually" update but with success

/Henrik

gudnimg, 2023-09-29:
@tbatur You are trying to flash the printer firmware on the MMU. **This will never work!** The MMU has it's own separate `.hex` file. It has always been like this.

Use this MMU firmware with 3.13.0:
[image]

For 3.13.1 and 3.13.2 you will need MMU FW 3.0.1

tbatur, 2023-09-29:
> Looks like you try to update the MMU with the printer firmware. You need to download the [separate hex [...]

It's done now. I didn't try it because the name of that file only says MMU3, thank you.

tbatur, 2023-09-29:
> @tbatur You are trying to flash the printer firmware on the MMU. **This will never work!** The MMU has it's own separate `.hex` file. It has always been like this.
> 
> Use this MMU firmware with 3.13.0: [image]
> 
> For 3.13.1 and 3.13.2 you will need MMU FW 3.0.1

Yes, it happened with him, but they should fix the name of that file. Because of the name it looks like it's only for MMU3.

gudnimg, 2023-09-30:
Closing as resolved. `MMU2S` will be added to the filename in future releases (https://github.com/prusa3d/Prusa-Firmware-MMU/pull/307).

closed 2023-09-30 by gudnimg