prusa3d/Prusa-Firmware #4424: Bad translation or bad display of the message

LoakAudio, 2023-10-02:
**Printer type** - MK3S+ no MMU
**Printer firmware version** -  3.13.1- French language selected

**Describe the bug**
 During a filament change, when it's time to introduce the new filament the message showed is not clear, I think last letters of the message are mixed at the start of it ( see photo)

[image]

In addition, when introduced, the filament loads automatically, no need to push a button as it seems to be said on the screen
  
**To Reproduce**
Try a Print with firmware 3.13.1 in french, no MMU, with a filament change 

  
**Expected behavior**
 A showed message describing which action is needed
  
Don't hesitate if you need more information
Thanks !

3d-gussner, 2023-10-03:
@LoakAudio Thanks for reporting the issue. Yes in French the line is too long `et appuyez sur le bouton` and isn't shown correctly on LCD screen
```
 |01234567890123456789|
0|
1|Inserez le filament |
2|et appuyez sur le bo|uton
```
And the `uton` overwrite part of the `Inserez`

3d-gussner, 2023-12-19:
@LoakAudio Merci beaucoup again I created a fix in #4543 and it looks like this
[image]

I am closing this issue. If you disagree please reopen it again.

closed 2023-12-19 by 3d-gussner