prusa3d/Prusa-Firmware #4425: [BUG] print fan is not uniformly

bigrip, 2023-10-04:
**Printer type** - MK3S+
**Printer firmware version** - 3.12.2

**Describe the bug**
 I swapped my print fan for a more powerful one. The print fan doesn't rotate evenly, the speed changes, it gets faster then slower again every 5 seconds.
When the fan runs at 100% this problem does not occur.
  
**To Reproduce**
set the printfan lower then 100%
  
**Expected behavior**
the printfan should always run at a consistent speed

3d-gussner, 2023-10-04:
The 100% fan burst are needed to get a clear rpm signal to ensure that the print fan turns. The PWM fan control distorts at different speeds the RPM signal line.

3d-gussner, 2023-10-04:
> is it possible to turn this off? With the part cooling fan it is not so critical if it does not rotate. I would even recompile the firmware if necessary.

Yes you can modify the firmware to ignore it (open source) but it isn't easy like a comment out of a `#define FANCHECK` as it will also NOT check the hotend fan and so potential heat creep. 
You gonna need to do more I guess.

3d-gussner, 2023-10-06:
Closing this issue. @bigrip If you don't agree with closing it feel free to reopen it.

closed 2023-10-06 by 3d-gussner