prusa3d/Prusa-Firmware #4431: Thermal Anomaly 3_13_1 MK3S+

Postr70, 2023-10-06:
### Description of the bug

With Firmaware 3_13_1 the Nozzle get "Thermal Anomaly" on my MK3S+
It's not possible to do a PID-calibration. 
Works fine with firmware 3_11_0 but now newer that that.
I use original Prusa Parts so no mods.

### Project file & How to reproduce

no file

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.1

### Operating system

Windows 10

### Printer model

Prusa i3 MK3S+

jwtoii, 2023-10-08:
I've had a similar problem with a Revo hot-end but I also can't run the terminal model calibration

cqeta1564, 2023-10-09:
Just turn it off with gcode M310 S0 and M500 for save.

acothebraco, 2023-11-12:
I noticed that a Thermal Cal. When using a housing (Prusa Enclosure, etc.), different values ​​are written into the EEPROM (Cold Enclosure, warm Enclosure), which then leads to the THERMAL ANOMALY error. This means that values ​​from the HOTEND and the surrounding area are taken into account. I'm not sure whether this can be updated with future firmware updates or at least that external factors such as ambient temperature don't play a role.

3d-gussner, 2024-01-02:
@acothebraco As the EINSY Ambient thermistor is used as a base for the Thermal model cold and warm TM cal may cause your reported issues.

Can you please measure the reported ambient temperature and place a "real" temperature near the einsy box/ in enclosure?
The default `THERMAL_MODEL_Ta_corr` is set to [-7°](https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/variants/MK3S.h#L416) for an open environment, where the heat of the bed and hotend don't change it a lot. 

1. Measure with Cold enclosure and cold printer
2. Start logging the temperature with `M155 S1` the output should be something like below T:hotend B:bed P:PINDA A:Ambient
```
T:16.0 /0.0 B:16.1 /0.0 T0:16.0 /0.0 @:0 B@:0 P:16.1 A:16.9
```
3. Open Support -> Temperatures to see the values on LCD 
4. Measure inside the enclosure with an additional thermistor the temperature.
5. Take some notes with time stamp, T, B, P, A and external temp at the same time
6. Start to preheat and a print and repeat the Step 5 every few minutes to see how the Ambient and enclosure temp change over time and how far these drift.
7. Try to get your setup to the temp it operates mostly
8. Repeat the Steps 2-7 above with a warm enclosure and warm printer.

It would be also great if you could provide detailed Thermal Model calibration data when executed in cold and in warm enclosure.
```
M310 S0
M155 S1 C3
D70 S1
M310 A F0
```

We suspect that the `THERMAL_MODEL_Ta_corr` may be the issue you have in the enclosure and getting some data and feedback from you would help to investigate the issue and work on a solution.

Another potential issue is that the cold air while part cooling causes the TM errors (when TM cal was done when warm) as it draws more thermal heat from the heat block (thermistor) than calculated during TM calibration. And vice versa.

acothebraco, 2024-01-07:
Hello @3d-gussner 

Thank you for the Mail.

I tired the steps that you asked for and now I provide the information to you in hope that we can found a solution for the problem. Just for your information, since I deactivated the Thermal Protection I don´t have any error messages. The Prints are also very fine.  The pictures attached below are in the same order like the output from Terminal.

Cold Printer:  T:31.9 /0.0 B:30.2 /0.0 T0:31.9 /0.0 @:0 B@:0 P:0.0 A:37.3
Ambient temp. on Display 37C 

**Preheating for Print Prusament PETG**

Recv: T:240.2 /240.0 B:84.7 /85.0 T0:240.2 /240.0 @:28 B@:52 P:0.0 A:42.2
Recv: E0:4380 RPM PRN1:0 RPM E0@:255 PRN1@:0
Recv: T:240.0 /240.0 B:84.7 /85.0 T0:240.0 /240.0 @:31 B@:59 P:0.0 A:42.0
Ambient temp. on Display 42C

**Printing Calibration Cube Prusament PETG**

Recv: T:249.8 /250.0 B:90.1 /90.0 T0:249.8 /250.0 @:58 B@:73 P:0.0 A:45.4
Recv: T:249.9 /250.0 B:90.1 /90.0 T0:249.9 /250.0 @:48 B@:84 P:0.0 A:45.8
Recv: T:250.2 /250.0 B:90.1 /90.0 T0:250.2 /250.0 @:51 B@:75 P:0.0 A:45.9
Ambient temp. on Display 46C

**Printing Calibration Cube Prusament PETG after 15min.**

Recv: T:249.8 /250.0 B:90.0 /90.0 T0:249.8 /250.0 @:55 B@:79 P:0.0 A:49.7
Recv: T:250.1 /250.0 B:90.0 /90.0 T0:250.1 /250.0 @:50 B@:75 P:0.0 A:49.8
Recv: T:249.8 /250.0 B:89.9 /90.0 T0:249.8 /250.0 @:55 B@:91 P:0.0 A:50.0
Ambient temp. on Display 50C

**End of the Print**

Recv: T:249.2 /250.0 B:89.9 /90.0 T0:249.2 /250.0 @:66 B@:84 P:0.0 A:52.9
Recv: T:250.8 /250.0 B:90.0 /90.0 T0:250.8 /250.0 @:41 B@:71 P:0.0 A:52.9
Recv: T:250.7 /250.0 B:90.0 /90.0 T0:250.7 /250.0 @:43 B@:66 P:0.0 A:53.0
Ambient temp. on Display 53C

[image]
[image]
[image]
[image]
[image]

**Thermal Model calibration Data** attached in the .txt File
[Thermal Calibration Data.txt](https://github.com/prusa3d/Prusa-Firmware/files/13851764/Thermal.Calibration.Data.txt)

3d-gussner, 2024-01-08:
@acothebraco Thanks for the detailed testing, very appreciated.
I  will look more into this issue but I can already tell that our suspicion about the `THERMAL_MODEL_Ta_corr` may be the issue why you get TM errors even after calibrating the TM correctly.

Your Enclosure thermistor temp vs Einsy Ambient temp values:
| State | E-Temp | A-temp | Diff /Ta_corr|
| ------- | ----------- | ----------- | ----------------- |
|Cold  | 31.0°C    | 37.0°C  | 6.0°C             | 
|Preheating  | 30.5°C    |42.0°C  | 11.5°C             | 
|Printing  | 31.2°C    | 45.9°C  | 14.7°C             | 
|Printing 15 min  | 32.2°C    | 50.0°C  | 17.8°C             | 
|End of print | 33.9°C | 53.0°C | 19.1°C|

As we can see the `-7°C` Temperature Ambient correction values is in cold state very close to what you have measured in the enclosure environment of `6°C`.
But during the print the difference between environment raises and raises. The Einsy heats up more than the outside of an enclosure, where it stays quite stable around the `7°C` higher than the environment.

acothebraco, 2024-01-08:
Thank you for the Information. I will wait for a solution or ist it maybe better to put the Einsy board outside of the Prusa enclosure?

3d-gussner, 2024-01-08:
> Thank you for the Information. I will wait for a solution or ist it maybe better to put the Einsy board outside of the Prusa enclosure?

Putting the EINSY outside the enclosure will not fix the issue as the TM relays on the EINSY Ambient temperature as as basis to compare the TM calculated values.
Also you would need extension cables for the motors, thermistors, etc.

acothebraco, 2024-01-08:
Ok than I leave temporary the TM off and when a fix in the future firmware will be ready that will fix the problem. Thank you

3d-gussner, 2024-01-08:
@acothebraco I understand that you disable the TM, but glad you said "temporary". :crossed_fingers: We find a good solution for the TM in an enclosure.

acothebraco, 2024-01-08:
> @acothebraco I understand that you disable the TM, but glad you said "temporary". 🤞 We find a good solution for the TM in an enclosure.

Thank you again. I´m proud and happy every time with the Prusa Support. keep the good work. 

Best Regards 

Aleksandar

acothebraco, 2024-03-24:
@3d-gussner any news about a fix? I’m now on the way to upgrade my MK3S+ to the MK3.5. Maybe the THERMAL ANOMALY problem will be solved with the upgrade with the new xBuddy Board?

3d-gussner, 2024-03-25:
@acothebraco The MK3.5 and all other buddy firmware based printer don't have the Thermal protection thermal model yet.
I have no update yet about the issue within the enclosure.

closed 2024-10-15 by github-actions[bot]

brucec87, 2024-11-03:
I have a stock Mk3 for 7 years.  Since 3.14 upgrade I have been getting THERMAL ANOMALY message regularly but intermittently.  In looking at the temps, the Nozzle and Bed temps are rock solid at the expected values.  I have never had this message before.  How do I diagnose the source?

3d-gussner, 2024-11-04:
@brucec87 What was the previous firmware you were using? Did you run the Calibration -> Thermal model cal.?
When you connect your printer via USB with a serial terminal you should be able to see when it is reporting the TM warnings / errors.

User have found
- thermistors and cables that were broken at some locations
- heater cartridges over- or under-perform (too slow or way too fast heating)
- Some objects that are printed tend to blow back at the nozzle which cools it down.