prusa3d/Prusa-Firmware #4434: [BUG] German tune menu missing text

ulab, 2023-10-08:
**Printer type** - MK3S
**Printer firmware version** -  3.13.1

**Describe the bug**
[Users have reported](https://www.printables.com/group/3d-printing-community-germany-dXZjmRk/comments/1178058) that the German translation of the tune menu is not showing text for the two lines "Speed" and "Main" since version 3.13.0.
  
[image]

**To Reproduce**
Install 3.13.1, load German, start printing and check the "Feineinstellung" menu.
  
**Note**
The English menu does not have this issue:

[image]

sarusani, 2023-10-11:
Duplicate of https://github.com/prusa3d/Prusa-Firmware/issues/4384, https://github.com/prusa3d/Prusa-Firmware/issues/4406 & https://github.com/prusa3d/Prusa-Firmware/issues/4413.

Already solved in [3.13.2](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.13.2-RC1).

ulab, 2023-10-11:
Oh wow. I didn't find anything when I was looking for them. Closing this ticket.

closed 2023-10-11 by ulab