prusa3d/Prusa-Firmware #4435: [BUG] MMU2S abnormal behavior after upgrade to 3.0.0

alehtosalo, 2023-10-08:
HW setup: MK3S+ & MMU2S
FW: 3.13.0 (3.0.0), then downgrading to 3.12.2 (1.0.6)

**Describe the bug**
After upgrading to 3.13.0 (3.0.0), MMU2S stopped working. It behavies abnormally, the bearing drum and the selector are not synced and the bearing drum is not pressing down the filament against the gear. Bearing seems to be off, either in front of or back of the sweet spot and filament does not move.

I do not recall what was the firmware before upgrade, most likely I have not been ipdating at least in a year or so. I downgraded back to 1.0.6 release (3.12.2. for printer) but no help. Same behaviour continues, it feels like the SW broke the HW some how. 

Video show manual movement of the selector and you can see that the drum does not move properly and it's lagging at least one position. MMU2S does not find correct slot and when it sometimes accidentally stops at right place, drum bearing does not add enough force to move the filament.

Tightening screws are not the issue, tried loosening and tightening them again. Either is filament. Tested with several types and manufacturers.

What is the working FW I need to flas to fix this? There is 1500€ of HW unusable because of this "upgrade". Apparently you make your customers test your releases and not do your job properly.
  
**To Reproduce**
Unknown. I just flashed the new SW and after fighting with that s**t for couple of days, downgtaded back to old one with no help.
  
**Expected behavior**
MMU2S should select correct filament

**Video**

https://github.com/prusa3d/Prusa-Firmware/assets/42060461/363167bc-be2a-48d4-9d8f-ac1c2a39d610

https://github.com/prusa3d/Prusa-Firmware/assets/42060461/ac0c98c2-6720-4cf9-b156-e2c69f9b8e1e

alehtosalo, 2023-10-11:
No video showing that there is more in this. When I use my hand to give enough pressure to bering so that filament can move, it goes to sensor and immediately it's whitdrawn. The sensor just blinks and the gear rotation direction changes.

https://github.com/prusa3d/Prusa-Firmware/assets/42060461/f97b96f8-aff3-47d1-ae6e-ee403a715f9c

alehtosalo, 2023-10-15:
1wk and no comments from Prusacrew...is this how you treat paying customers??? You so called SW upgrade destroyd my printer, PLEASE proceed with fix for this ASAP!!!

gudnimg, 2023-10-15:
I can’t speak on behalf of Prusa Research but want to say this isnt the platform to get official support especially when your problem seems like its hardware related. I would go through Prusa Research’s website for that and contact Technical Support. 

The videos you shared don’t look like MMU firmware 3.0.0? It moves way too slow like the old MMU FW 1.0.6. You’re certain the update went through?

The firmware is fully tested and should be working just fine out of the box. So whatever is happening with your unit is not a common problem. 

Does the printer show any MMU error messages that may shed light on the problem?

alehtosalo, 2023-10-15:
Problem started with 3.0.0 upgrade and after that I downgraded to 1.0.6 moving down with printer FW. Videos have been taken with 1.0.6 yes, but problem started with the updage. This morning flashed 3.1.0.0 and re-flashed also 1.0.6 with factory reset. No help. So basically if problem is HW related, FW 3.0.0. broke something in my printer or MMU. Before that, it worked well for over 1,5years.

gudnimg, 2023-10-15:
@alehtosalo did you downgrade the printer as well? You can't use 3.13.0 printer FW with old MMU FW.

alehtosalo, 2023-10-15:
> @alehtosalo did you downgrade the printer as well? You can't use 3.13.0 printer FW with old MMU FW.

Yes, of course. SW package has FW for both. Also factory reset has been done for the MMU every downgrade.

closed 2024-01-04 by github-actions[bot]