prusa3d/Prusa-Firmware #4436: [BUG] Continuous Beep Noise with MINTEMP Error on Firmware 3.13.1

dalesd, 2023-10-09:
**Printer type** -. MK3S+
**Printer firmware version** - 3.13.1

**SD card or USB/Octoprint**
Octoprint

**Describe the bug**
After updating from firmware version 3.10 to 3.13.1, my printer, located in a cool unheated basement with temperatures ranging from approximately 10-15°C depending on the season, exhibits an issue with the MINTEMP error. Specifically, when the MINTEMP error occurs, the printer emits a loud, constant beep noise, even when the printer is in an idle state with no active printing or heating processes. This loud and continuous alarm noise persists until I manually power down the printer.
  
**To Reproduce**
1. Update the printer firmware to version 3.13.1.
2. Place the printer in a cool unheated basement environment (10-15°C).
3. Allow the printer to reach the MINTEMP error state, for instance, by disconnecting or simulating a temperature sensor issue.
4. Observe the continuous and loud beep noise even when the printer is in an idle state.
  
**Expected behavior**
Prior to the firmware update to version 3.13.1, the MINTEMP error did not result in the continuous beep alarm noise when the printer was in an idle state. I expect the printer to maintain a quiet state when it is not actively in use or heating, even in the presence of a MINTEMP error.
  
**G-code**
N/A

## Impact:
The continuous beep alarm noise is highly disruptive and unacceptable in a household setting, as it annoys everyone present. Additionally, it renders the use of firmware version 3.13.1 impractical, as the alarm noise continues to sound after a print job is completed and the printer cools down.

## Additional Information:
I would like to understand the necessity of such an intrusive alarm when the printer is idle, with all heaters switched off. This alarm behavior was not observed in previous firmware versions, and it significantly affects the usability of the printer in a residential environment.

gisforgirard, 2023-11-19:
https://github.com/prusa3d/Prusa-Firmware/issues/3053

rongith, 2023-11-27:
Sorry to not bringing something constructive but having 5 enclosed and idled printers continuously beeping is quite annoying.
I very hope there will be a correction very soon.

makerspet, 2024-01-10:
I have exactly the same problem after upgrade to 3.13.2 with two MK3.5S, in an unheated garage when the temperature drops under ~10C at night.
Will you please fix this? The continuous beep sound from both printers is incredibly annoying!

gisforgirard, 2024-01-26:
`diff --git a/Firmware/temperature.cpp b/Firmware/temperature.cpp
index cc052ce0..37187ab9 100755
--- a/Firmware/temperature.cpp
+++ b/Firmware/temperature.cpp
@@ -1446,7 +1446,7 @@ void check_min_temp_heater0()
 #else
 	if (current_temperature_raw[0] <= minttemp_raw[0]) {
 #endif
-        set_temp_error(TempErrorSource::hotend, 0, TempErrorType::min);
+        //set_temp_error(TempErrorSource::hotend, 0, TempErrorType::min);
 	}
 }
 
@@ -1457,7 +1457,7 @@ void check_min_temp_bed()
 #else
 	if (current_temperature_bed_raw <= bed_minttemp_raw) {
 #endif
-        set_temp_error(TempErrorSource::bed, 0, TempErrorType::min);
+        //set_temp_error(TempErrorSource::bed, 0, TempErrorType::min);
 	}
 }
 
@@ -1469,7 +1469,7 @@ void check_min_temp_ambient()
 #else
 	if (current_temperature_raw_ambient <= ambient_minttemp_raw) {
 #endif
-        set_temp_error(TempErrorSource::ambient, 0, TempErrorType::min);
+        //set_temp_error(TempErrorSource::ambient, 0, TempErrorType::min);
 	}
 }
 #endif
@@ -1479,6 +1479,7 @@ void handle_temp_error()
     // relay to the original handler
     switch((TempErrorType)temp_error_state.type) {
     case TempErrorType::min:
+	/*
         switch((TempErrorSource)temp_error_state.source) {
         case TempErrorSource::hotend:
             if(temp_error_state.assert) {
@@ -1508,6 +1509,7 @@ void handle_temp_error()
             break;
 #endif
         }
+	*/
         break;
     case TempErrorType::max:
         switch((TempErrorSource)temp_error_state.source) {
`

1. clone the Prusa-Firmware repository
2. save this to a file named mintemp.patch within the Prusa-Firmware directory
3. using the command line, cd to the Prusa-Firmware directory and type `git apply mintemp.patch`
4. print outside without wanting to throw the printer in the nearest lake

closed 2024-04-03 by github-actions[bot]