prusa3d/Prusa-Firmware #4444: First layer calibration not working when filament sensor is switched off

ArjanStr, 2023-10-12:
### Description of the bug

First layer calibration not working when filament sensor is switched off.
When the menu is asking for 'filament loaded?' both options yes and no do not forward the menu
Prusa MK3, prusa3d_fw_MK3S_MK3S+_3_13_1_6876.hex firmware.
When the filament sensor is switched on, all is working ok

### Project file & How to reproduce

Do first layer calibration with filament sensor switched off

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

prusa3d_fw_MK3S_MK3S+_3_13_1_6876.hex

### Operating system

windows 10

### Printer model

Prusa MK3

frogale, 2023-10-21:
I can confirm that.. I am using Revo edition 3.13.1 and cannot proceed with first layer calibration.. filament sensor off..

gudnimg, 2023-10-21:
@frogale Fix is coming in 3.13.2, you can use https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.13.2-RC1 if you want, but the final 3.13.2 should come next week with some minor changes on top of RC1.

gudnimg, 2023-12-21:
Closing since this was fixed in 3.13.2

closed 2023-12-21 by gudnimg