prusa3d/Prusa-Firmware #4445: Back button on MK3s+

joker3d, 2023-10-12:
Hello, is possiblee to make back button on MK3S+ with MMU2/3? 
When i click start to print selected item, printer will display "choose extruder 1-5"
but... there is only one option how to get back... reset printer....
for me mk3s+ have slow rotation button and sometime missclick and select wrong line. 
in menu "select extruder" is not possible to go back and choose correct one... 
Thanks Michal

autonumous, 2023-10-13:
I second this.  
Had exactly the same issue today a couple of time today,  eg selected 'cut filament' but accidentally selected the wrong one, one with no filament, there is no way of getting back without reseting.   There should be a back/cancel option in addition to retry.

cozmo14047, 2023-10-13:
I too want this when I select what fillament to load sometimes I click the wrong one maybe a long press of the knob

Elcar903, 2023-10-21:
I agree most menus need back button, some have them but not all and I have had to hit the "X" button a few times over the years because of miss click or something else.

gudnimg, 2023-10-22:
I don’t think this is very trivial to do with the current infrastructure. But one solution I can see is to add a confirmation dialog e.g. “Load filament 3?” And then “Yes” or “No”

If “No” then show the selection menu again. 

It’s very risky IMO to add a back button and allow users to leave the selection menu alltogether, at least in the M600 g-code I’m not sure how to handle that.

A problem with the MMU code is we can’t cancel a load if it has been started. At least not today. So adding a Back/Cancel button there is not possible. 

This issue isnt only limited to the MMU Load filament menu… I have experienced it also when selecting a file to print and all the LCD menus.

cozmo14047, 2023-10-26:
Yeah a confirmation menu for selecting fillament will do. Please add it

cozmo14047, 2023-12-28:
Any updates on this?

gudnimg, 2023-12-29:
@cozmo14047 No update yet.

closed 2024-10-16 by github-actions[bot]