prusa3d/Prusa-Firmware #4446: [BUG] Thermal model calibration impossible with RPi Zero PrusaLink connected

jkavalik, 2023-10-13:
**Printer type** - MK3S
**Printer firmware version** -  3.13.1

**MMU upgrade** - MMU2S
**MMU upgrade firmware version** - 3.0.0

**SD card or USB/Octoprint**
  PrusaLink on RPi Zero 2

**Describe the bug**
  Thermal Model Calibration does not run when PrusaLink is connected to the printer.
  It prints "TM: Cal. NOT IDLE" on the display after homing and does not continue.
  It was possible to run the calibration after disabling the RPi Port setting.
  The printer display shows "USB: ---" line so it probably counts as "active"
  
**To Reproduce**
  - printer with working and initialized Prusalink installed on "embedded" Raspberry Pi Zero
  - run Thermal Model Calibration

gudnimg, 2023-10-14:
Did you send some G-code commands to the printer before? The printer indeed shows "`USB: ---"` if it thinks a print is ongoing. What made it active in your case and when does it start?

I am unable to reproduce this with Octoprint. But can't rule out it only applies to PrusaLink yet.

jkavalik, 2023-10-14:
@gudnimg: just tried again, the USB actually jumpa on the display after homing just before the error message, when the extruder is moving from the front left to the center of the bed. But one thing that "might" be a part of it, my mmu2 with the v3 firmware has some problems with homing the idler, it does error after turning on but then works after a reset (by the X button), at that point in time the RPi and PrusaLink is already booted and is not reset again, only reconnects to the printer, so that might be it.

gudnimg, 2023-10-14:
Hmm, I'm also using MMU3 and don't see any issue.

How are you starting the thermal calibration? Do you go to `LCD menu -> Calibration -> Thermal Model cal.` using the LCD knob? Or are you doing it through G-code?

jkavalik, 2023-10-14:
No, MMU is not it, I disabled it in the menu to not have to reset after startup and the problem persists.

Yes, in the Calibration menu.

3d-gussner, 2023-10-16:
@jkavalik Thanks for reporting this issue. I have been able to reproduce it.
I started PrusaLink in debug mode for the serial output and think I have found the cause.

PrusaLink (PL) is constantly monitoring the  printer and when `G28`  is seen by PL, it sends a  `G81` to get the MBL table.

Any `Gxx` gcode sent on serial will switch the printer to USB mode and sets the printer state to `active`.

The Thermal Model calibration is very compute sensitive and doesn't work correctly when other things happen, this why it doesn't proceed with the calibration as long the printer is `active`.

For FW 3.14.0 we introduce [M420](https://github.com/prusa3d/Prusa-Firmware/pull/4396) for PL to NOT trigger the USB host mode.

I will discuss with the PrusaLink developers if we can avoid this issue until FW 3.14.0.

For the time being please disable PrusaLink when running `Thermal model cal.`

jkavalik, 2023-10-16:
@3d-gussner Thank you for looking into it. Luckily the workaround is simple enough :)

antimix, 2023-11-10:
@3d-gussner Keep in mind if you have an MMU3 you are bombarded by diagnostic status message from the MMU3 and it does not work either. I was even unable to issue a simple MMU3 register command through serial USB mode because it was really impossible to write with tons of message arriving on console.
So, regarding REVO I had to disable the MMU3 to perform a PID calibration on the FW  3.13  :(

3d-gussner, 2023-11-21:
@antimix Yes the MMU reports lot of things on the serial line and makes it "hard" to read something.
When ever I use Ocotprint (mostly for testing) I added a terminal filter to reduce the terminal output on the screen.

There are few terminal tools that send the whole line and not char by char over the serial line.

Your workaround to disable the MMU is working as well.

3d-gussner, 2023-11-21:
@jkavalik Please consider to close this issue.

3d-gussner, 2023-12-19:
@TojikCZ How can you disable PrusaLink on RPi? see above

TojikCZ, 2023-12-19:
Ssh into it and run prusalink stop, Sudo killall python? Unplug the raspi, toggle the rpi port in the settings. Do any one of these

closed 2024-02-26 by github-actions[bot]