prusa3d/Prusa-Firmware #4447: [BUG]<baby bug: language spelling error>

ESF-nl, 2023-10-14:
Sorry to bother you with this small issue but my printer writing incorrect bothers me. In the language file for Dutch 
[https://github.com/prusa3d/Prusa-Firmware/blob/MK3/lang/po/Firmware_nl.po](url)
at line 229 there is a small language typo. The text "kalibrere" should be "kalibreren" the N is missing at the end.

sarusani, 2023-10-14:
Thanks for reporting!

3d-gussner, 2023-10-16:
@ESF-nl I think you have to read it in a context 

This message is shown when the printer encounters an issue and enforces an auto Z calibration.

It is more : I, your printer, start `calibrating Z` and not `Calibrate Z`

Ik denk `kalibrere Z` is niet heelmaal just, zou `Kalibreren Z` beter zijn? Nederlands is niet mijn moedertaal.

ESF-nl, 2023-10-16:
Yes, Dutch is my mothertongue and this was indeed my issue. Please change "kalibrere" into "kalibreren". The position of the axis name is not such an issue. This can be written as prefix or suffix. I'd propose to keep it in line with the orignal English version.

gudnimg, 2023-10-21:
Closing this issue because a fix has been merged into 3.14 development branch.

closed 2023-10-21 by gudnimg