prusa3d/Prusa-Firmware #4451: E3D REVO Thermal Model

3d-gussner, 2023-10-16:
1. Verify firmware version installed on your printer , to check this on the printers start screen the firmware version will be indicated you can also check which firmware version you already have via the following process L**CD menu -> Support. Scroll down** and you will see the firmware version.

If your printer displays a firmware version prior to 3.13.0 or it does not start with **E3DREVO_fw_** or **E3DREVO_HF_60W_** we would advise that you install the latest Revo supported firmware release via the following link and proceed with the next steps: https://github.com/prusa3d/Prusa-Firmware/releases
     
If you are using a E3D Revo firmware, please check that this matches the Revo Hotend you have installed as there are different versions.
- For a Standard 40W Revo heater core please use the. HEX file starting with **E3DREVO_fw_** ensuring you chose the correct. HEX file for the printer model you have.
- For a High flow 60W HeaterCore please use the. HEX file starting with **E3DREVO_HF_60W_** ensuring you chose the correct. HEX file for the printer model you have.

**If you do have an E3D Revo Firmware installed and it matches the Revo HotEnd you have please go to step 4.**

2. Once you have downloaded the firmware this Zip file and extracted it , please use the applicable. HEX file for both your printer model and the version of HeaterCore you have.
   - For a Standard 40W Revo heater core please use the. HEX file starting with **E3DREVO_fw_** ensuring you chose the correct. HEX file for the printer model you have.
   - For a High flow 60W HeaterCore please use the. HEX file starting with **E3DREVO_HF_60W_** ensuring you chose the correct. HEX file for the printer model you have.

_If you are unsure as to how to update your printers firmware with the .HEX file please refer to the Flashing procedure within the following guide:_ https://help.prusa3d.com/article/firmware-updating-
mk3s-mk3s-mk3_2227

3. Once you have flashed and updated your printers firmware Run both the PID and TM calibration from the LCD Calibration menus of your machine.
4. If step 2 fails however please Use the Revo_Reset G-code files to force setting changes. There are 2 versions of this g-code that will need to be used based on the HeaterCore you have installed.
    - **[reset_revo.gcode.txt](https://github.com/prusa3d/Prusa-Firmware/files/12917795/reset_revo.gcode.txt)**: for a 40W HeaterCore. Please rename the file form `reset_revo.gcode.txt` to `reset_revo.gcode` 
    - **[reset_revo_hf_60W.gcode.txt](https://github.com/prusa3d/Prusa-Firmware/files/12917816/reset_revo_hf_60W.gcode.txt)**:  for a 60W High Flow HeaterCore. Please rename the file form `reset_revo_hf_60W.gcode.txt` to `reset_revo_hf_60W.gcode` 

These files will need to be copied to an SD card and installed  , to Install the G-code select it from the print from the SD menu on your printer and follow the on-screen instructions.

5. Once the G-code has been installed re-run both a PID and TM calibration from the LCD Calibration menus of your machine.

**Optional step:** If the above steps fail you may wish to factory reset the machine and run the setup wizard. Please note that as the process suggests this will reset the printer and its firmware to a factory state and any setting will be lost.

[Link to E3D ticket](https://e3d-online.zendesk.com/hc/en-us/requests/new?ticket_form_id=11304553253917)

acothebraco, 2023-11-12:
Hello,

I have the same issute that my MK3S+ beeps three times with Thermal Anomaly and at the next time it stops with Thermal Error. I use the Revo Six Hotend and have never a problem with THERMAL ANOMALY. But since I put the MK3S+ in the Original Prusa Enclosure it shows up after 1-2 hours printing. I did also PID and Thermal Cal. on the 3.13.2 and the older 3.13.0 and I also tired to to a Revo_Reset G-Code without scuccess.  Only when i put the Printer outside of the Enclosure the Error will not apear. 

[image]

Lino77, 2023-11-17:
same problem for me. I updated from 3.13.0 to 3.13.2, set the printer to factory settings, I can't even do the first layer of calibration. without the printer screaming, revo six for mk3

Ezrae, 2023-11-17:
Adding my voice to this. I went through each step carefully. I reset TM and PID multiple times. I ran the gcode. And when I finally decided to factory reset, I was unable to finish the wizard due to the thermal errors I was continuing to get.

I've downgraded back to 3.11, and I'll try again with the next REVO firmware update.

nemith, 2023-11-25:
Two printers, did all the steps here. All had the same issue.  One seems to actually finish the model but still fails later, the other can't even calibrate the model.  

Going to downgrade back to 3.11 this feature seems half baked and not suited for revo hotends.

Lino77, 2023-11-25:
why downgrade , you can disable it.
[The thermal model protection can be disabled by connecting the printer to your computer (Octoprint, PuTTY...), and sending M310 S0. An additional M500 will save the setting and the Thermal model protection will remain disabled after the next restart.](https://help.prusa3d.com/article/thermal-model-calibration_382488#:~:text=model%20debug%20logging-,Disabling%20the%20Thermal%20model%20protection,the%20Thermal%20model%20protection%20will%20remain%20disabled%20after%20the%20next%20restart.,-Was%20this%20article)

nemith, 2023-11-25:
> why , you can disable it.

I am selling this printer and it need to "just work".  Disabling until next restart isn't an option.

Lino77, 2023-11-25:
M500 safe it and it will remain disable afer restart

LoakAudio, 2023-11-29:
@3d-gussner , In the step 4 of your first post, isn't it "if step 3 fails" rather than "step 2" ? because it appends to me : changing the firmware went fine, reboot OK with the new firmware but PID calibration fails and TM calibration fails too. The gcode saves me.

3d-gussner, 2023-11-29:
@LoakAudio Thanks for the feedback I will check again.
Glad that the new firmware and gcode files saved you. :tada:

jamesdehart, 2023-11-30:
@3d-gussner I happen to do a factory reset and after the calibration this happened

https://github.com/prusa3d/Prusa-Firmware/assets/6137749/24a2215f-4de8-4e8a-a89c-a9ace7ca4f83

3d-gussner, 2023-12-01:
@jamesdehart Which calibration caused this screen?

Skilbo, 2023-12-04:
After much head scratching, finally got it working, used the E3DREVO_fw version and everything worked first time. Thanks to @3d-gussner for laying out the procedures very much appreciated.

jamesdehart, 2023-12-11:
> @jamesdehart Which calibration caused this screen?

So it turned out my entire issue was because the thermal cable connector from the Hot-end to the cable that goes to the motherboard failed.

[image]

For others who are running in to issues related to this hotend. It would be good to check there cable to see if there are any issues like I had. When contacting E3D it took them about a week to respond to the support tickets. They are really backed up.

I hope this helps.

3d-gussner, 2023-12-12:
@jamesdehart Thanks for sharing! This is an example why we implemented the Thermal model to detect issues like this and other.
Well done to investigate :medal_sports: instead of disabling the TM!

acothebraco, 2023-12-12:
Thanos for sharing. I checked my printer six times and I don’t have any cable or hardware issues.  The point is that like I wrote in a other post that TM have problems when the printer is in a enclosure.

ndemarco, 2024-01-08:
I'm chasing down the same `THERMAL ANOMALY` trouble,

- Verified the wiring is correct and in working order
- run [reset_revo.gcode.txt](https://github.com/prusa3d/Prusa-Firmware/files/12917795/reset_revo.gcode.txt)
- recalibrated PID
- recalibrated TA
- hard reset, followed by the usual recalibrations from the wizard

The PID and TA calibrations finished properly. Upon the next PLA preheat, `BEEP-BEEP THERMAL ANOMALY`.

3d-gussner, 2024-01-08:
@ndemarco Thanks for the update.
Is your printer in an enclosure?
Can you please try to capture the print in a logfile? Do you have a host system like Octoprint running?

stevenbennett, 2024-01-12:
I'm also experiencing the issue described by users in this thread. Installed the Revo variant of 3.13.2 on two MK3S+ and now encounter consistent thermal anomaly errors. Things I've tried:

- verified heater/thermister cable and cable extension integrity
- ran reset_revo gcode mentioned in https://github.com/prusa3d/Prusa-Firmware/issues/4451#issue-1945318827
- performed full factory reset
- re-ran PID calibration
- re-ran thermal model calibration

After these steps I still encounter frequent thermal anomaly errors. Sometimes these occur right after initiating a preheat, sometimes right before a print begins, sometimes during a print. (I'm not using and enclosure). Going to try rolling back to 3.11 as other have suggested.

slaesh, 2024-02-01:
had the same issue. it works now, maybe change the order of your steps 2 and 3 @stevenbennett :)
1. update firmware
2. factory reset
3. ran the gcode magic
4. PID calibration
5. thermal calibration

seems to work, step 3 was the magic sauce for me though! ;)

3d-gussner, 2024-02-01:
> had the same issue. it works now, maybe change the order of your steps 2 and 3 @stevenbennett :)
> 
>     1. update firmware
> 
>     2. factory reset
> 
>     3. ran the gcode magic
> 
>     4. PID calibration
> 
>     5. thermal calibration
> 
> 
> seems to work, step 3 was the magic sauce for me though! ;)

After the factory reset you don't need to run any gcode as it set identical values as are used in the E3D . hex files https://github.com/prusa3d/Prusa-Firmware/blob/v3.13.2/Firmware/thermal_model/e3d_REVO.h

slaesh, 2024-02-01:
I am pretty sure I did the factory reset first. And then doing the PID
calibration it started and immediately stopped and was in a weird state
afterwards. Seems like I did the factory reset with the old "normal"
firmware though 🤔🙈. Sh*t. Thanks for the hint. will start all over again
with a factory reset now and see what happens! ;) Sorry for the confusion 🤫

jake-carpenter, 2024-03-11:
I just installed a REVO because it sounded like this has been sorted out, unfortunately I got errors immediately with v3.13.3-Revo.

* Printer: MK3S+. 
* Currently not in my enclosure, but I'll try it in the enclosure later.
* I'm doing my FW flashing from Octoprint, if that has any bearing.

Steps I have taken:

1. Flashed v3.11 and tried factory reset then PID calibration, as some old posts have recommended.
  a. Ran a print on v3.11 to confirm everything does work fine.
2. Flashed 3.13.3-Revo (7094).
3. Performed a full factory reset.
  a. I let the first-time setup run and calibrate everything. TM and PID were calibrated.
  b. At this point, prints would start but would not go for long before I got a thermal error.
4. Performed the Task 1 steps in #4105 (via Octoprint).
  a. Thermal error almost immediately after a print started.
5. Ran the `reset_revo.gcode` provided above from my SD card.
  a. Still have thermal errors within seconds of starting the print.

Not sure what else to do at this point. I'd be willing to re-check all the wiring for bad connections, except everything is fine on v3.11. In fact, it's printing on v3.11 as I write this. I'm happy to provide some logs if it will help here. I'll just need a link on what/how to pull them.

3d-gussner, 2024-03-12:
@jake-carpenter 

> I'm doing my FW flashing from Octoprint, if that has any bearing.

Please don't update your MK3/S/+ firmware with OctoPrint firmware flasher. It is still missing the extension for the MK3/S/+ language packs https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/pull/325

Ensure that you flash the correct firmware version for the E3D Revo which is in the https://github.com/prusa3d/Prusa-Firmware/releases/download/v3.13.3/e3d_REVO_fw_MK3_MK3S_MK3S+_3_13_3_7094.zip
Looks like you did it in Step 2

The `reset_revo.gcode` writes the default E3D Revo TM and PID values to eeprom so you don't have to factory reset the printer. When you run this gcode you need to run PID and TM calibration again to get values for your printer.

Please also contact E3D Support for more help https://e3d-online.zendesk.com/hc/en-us/requests/new?ticket_form_id=11304553253917 check also their discord.

You can enable in OctoPrint save logfiles to get more information during the print. These logfiles should help E3D support to narrow down your issue.

IndustrialCADWorks, 2024-04-02:
I just had this problem. Went back, went through everything listed above and other forums. What I might have figured out is on step 40 on the E3D instructions.

https://e3d-online.zendesk.com/hc/en-us/articles/4410028113169-Revo-Six-Prusa-MK3-MK3S-MK3S-upgrade-guide

It mentions PID tuning at 240 degrees if you plan to use materials other than PLA. When running the PID calibration through the wizard it runs at 230. I went back Calibrated PID to 240, ran through all the steps and no thermal issues or haults through the calibration. So I'm assuming you have to run the PID calibration prior to running the configuration wizard?? I don't know if that is true, but I know it worked for me.  Hope this helps.

mhouse1, 2024-04-11:
> I just had this problem. Went back, went through everything listed above and other forums. What I might have figured out is on step 40 on the E3D instructions. [...]

Does it still work after a week? In my experience, it'll work a few prints but eventually it starts erroring out during the middle of a print, you can restart and print again but with the risk of failure we reverted to 3.11

IndustrialCADWorks, 2024-04-11:
I've been printing everyday and no problems, (pla, PE / CF, Nylon, PETG) multiple nozzle changes, no issues so far. Like I said not sure if it helps, but just posting my experience. If I have any issues I'll have to revert to 3.11, but for now no problems so far. Fingers Crossed...

3d-gussner, 2024-04-24:
@ndemarco This is a very detailed report and video, really appreciated. Makes troubleshooting sooo much easier.

Please try to connect your RPi (guess OctoPrint) enable serial logging in Octoprint and connect to the printer. Serial logs help a lot while troubleshooting.

You can use the https://github.com/prusa3d/Prusa-Firmware/files/12917795/reset_revo.gcode.txt (change filename extension to gcode) to reset to the default REVO PID and Thermal Model values.

To fine tune the PID and REVO on your machine
1. Let it cool down
2. LCD Calibration -> PID calibration
3. Let it cool down
4. LCD Calibration -> Thermal Model cal.

If that doesn't help please do a factory reset.

In your video you do a Factory reset BUT you only reset the language. You need to scroll down to "All Data" to reset everything.
Keep in mind that it will delete your statistics.

The comment from https://www.youtube.com/@n.neves1776 is good but only needed if you don't get it working with the steps above.

Why it may have failed on your printer:
I think that you
- had Thermal Model calibrated for the E3Dv6, which is stored in EEPROM
- have the PID values for the E3Dv6, which are stored in EEPROM
- then you changed to the E3D Revo and tried to run Thermal Model cal., which runs the TM calibration taking into account the previous stored values in EEPROM.
  - As the previous EEPROM values are for the E3Dv6 and these differ a lot from the E3D Revo the TM cal will complain that these outside expected thresholds and report the TM ERROR.

3d-gussner, 2024-04-24:
@ndemarco I can understand the frustration, but the Thermal model saved already few users with broken cables and connectors.
The TM runs a simulation based on the calibrated values and compares it to the real data of the thermistor and power provided to the heater cartridge, so if the TM and PID values are from the E3Dv6 and the real data is from REVO it will report TM warnings and Errors.

ndemarco, 2024-05-31:
@3d-gussner I appreciate all the feedback. I am carefully trying each of your points. In this video, I show the firmware flashing (which fails the first time around and succeeds the 2nd time). I then show the whooooooole setup, so you may want to >> for some. When it tries to TM calibrate, everything fails - STOPPED.

Now, I am waiting for the nozzle to cool completely. I will then run the reset_revo.gcode, followed by your [steps](https://github.com/prusa3d/Prusa-Firmware/issues/4451#issuecomment-2075305893) above - doing the PID calibration first.

ndemarco, 2024-06-02:
The PID calibration completed normally. Unfortunately, the TM calibration fails within a few seconds of starting.

mhouse1, 2024-06-02:
fwiw 3.11 is the  last stable version without the new thermal model, and really seems to be the only way forward if you want to use the much better revo six extruder.

Sp00k1Gh0st, 2024-06-05:
Hi everyone I just wanted to share some information I found while diagnosing the thermal anomaly issue I was having with the revo. With mine I was finding that when the anomaly would happen I would have to disconnect the hot end from both jump wires and reconnect them before running the code provided above to reset everything successfully. I have found that a connection break is happening at the heater connection but not at the thermostat connection. After I cut the ends off of the jumper and hot end and soldered the wires together I was able to eliminate the problem all together and have been printing for over a week now without issue. I am sure there is more going on but hopefully this helps!

3d-gussner, 2024-06-06:
@Sp00k1Gh0st Thanks for sharing the information.
The thermal model triggered here correctly and you did the right thing: Look why it triggers and not just disable it. :clap: :medal_sports: 

As you can see the thermal model is not to annoy users but to improve the safety of the hotend, which can get very very hot. It detected the broken/loose/unreliable connection and @Sp00k1Gh0st investigated the issue to solve it.

This is another report I have seen that some connectors may cause issue. But it isn't only a REVO thing, we also have seen that some wires broke somewhere in the loom and only were "faulty" at specific positions or movements.

ndemarco, 2024-06-06:
> @Sp00k1Gh0st Thanks for sharing the information. The thermal model triggered here correctly and you did the right thing: Look why it triggers and not just disable it. 👏 🏅
> 
> As you can see the thermal model is not to annoy users but to improve the safety of the hotend, which can get very very hot. It detected the broken/loose/unreliable connection and @Sp00k1Gh0st investigated the issue to solve it.
> 
> This is another report I have seen that some connectors may cause issue. But it isn't only a REVO thing, we also have seen that some wires broke somewhere in the loom and only were "faulty" at specific positions or movements.

This is most definitely not the issue with my Prusa + Revo printer. I've gone through the wiring and connections way more carefully than the Apollo 1 technicians did.

acothebraco, 2024-06-06:
Like I say. I will no longer wait for prusa to solve my thermal problems with the MK3S+ in a enclosure. I sell my mk3s and switched to another printer.

mhouse1, 2024-06-11:
I will give these [steps](https://github.com/prusa3d/Prusa-Firmware/issues/4451#issuecomment-2075305893) another try  once 3.14.0 hex is released; it was tagged 3 weeks ago but not built as hex yet. 

I will also try replacing the wires to the extruder if errors occur.

ndemarco, 2024-06-12:
Latest update:

- Swapped Einsy Rambo board from a known good MK3
- Verified all wiring 2x
- Flashed firmware file E3DREVO_fw_MK3S_MK3S+_3_13_3_7094.hex
- Factory reset all (more than language)
- Begin wizard
- Observe all tests pass
- Observe 6+ beeps followed by 12 beeps.
- Observe small circle in lower left, with center dot.

Point me in a direction to solve this please.

mhouse1, 2024-06-12:
the instructions https://github.com/prusa3d/Prusa-Firmware/issues/4451#issuecomment-2075305893 above isnt  clear but i assume you need to repeat these calibration steps after a factory reset; which you didnt do in your video, and im interpreting it as you dont need  to run the [reset_revo.gcode.txt](https://github.com/prusa3d/Prusa-Firmware/files/12917795/reset_revo.gcode.txt)  if you do the factory reset. 

```
To fine tune the PID and REVO on your machine

Let it cool down
LCD Calibration -> PID calibration
Let it cool down
LCD Calibration -> Thermal Model cal.
```

ndemarco, 2024-06-12:
> the instructions #4451 (comment)

I don't even get to the point of running PID or TM. I get a series of beeps
and an o.

gouryellas, 2024-06-15:
> Latest update, [captured on video](https://youtu.be/-1ImEIG6ytw):
> 
> * Swapped Einsy Rambo board from a known good MK3
> * Verified all wiring 2x
> * Flashed firmware file E3DREVO_fw_MK3S_MK3S+_3_13_3_7094.hex
> * Factory reset all (more than language)
> * Begin wizard
> * Observe all tests pass
> * Observe 6+ beeps followed by 12 beeps.
> * Observe small circle in lower left, with center dot.
> 
> Point me in a direction to solve this please.

downgrade to 13.11 until they figure things out. been using it for a long time with no problems on the revo

Leopere, 2024-06-27:
Is this fixed in the latest version as of today?

ndemarco, 2024-06-27:
I can do some testing over the weekend.

Leopere, 2024-06-28:
I haven't had a thermal alert yet but I'm not sure what caused my first big one.

ndemarco, 2024-07-01:
> I will give these [steps](https://github.com/prusa3d/Prusa-Firmware/issues/4451#issuecomment-2075305893) another try once 3.14.0 hex is released; it was tagged 3 weeks ago but not built as hex yet.
> 
> I will also try replacing the wires to the extruder if errors occur.

@mhouse1 Have you tried the RC 3.14.x yet? If you had good results, I'll also try it and report back.

mhouse1, 2024-07-01:
> @mhouse1 Have you tried the RC 3.14.x yet? If you had good results, I'll also try it and report back.

3.14 is tagged but there are no hex files  included so I cant use it yet. https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.14.0

Leopere, 2024-07-01:
Sounds close

mhouse1, 2024-07-02:
new firmware is ready, i will test this on July 4th and report back, looking forward to getting this resolved and especially the new reprint menu feature; it will really help with running the farm of 100's of files.

mhouse1, 2024-07-10:
> new firmware is ready, i will test this on July 4th and report back, looking forward to getting this resolved and especially the new reprint menu feature; it will really help with running the farm of 100's of files.

So far its working great on the 3 printers I updated a week ago; fw3.14 for mk3s and mk3s+ with revo.

 I think the main difference was running the reset_revo.gcode , I didnt have to factory reset any printer. I tried this on printers as old as 2019 with 982days of print time and new printer built in late 2023 with 40days of print time. 

The older printer starts thermal model cal right away after firmware update, newer one didnt. printer number 49 was stuck at PID cal 0/5 but I just restarted the printer and tried again.

Details:
```
====================================
printer #50 built 9/10/23
flash new mk3s revo hex firmware 3.14
shows prusa i3 MK3S+R ok on LCD

printer  statistics total filament 2766.26m
total print time 40d

use reset_revo.gcode on sd card
response: file incomplete continue anyway?
yes
Press to reset REVO
Press to Save
0 hours 1 minutes

next, performed PID  calibration to 250

thermal model calibration
TM: initial C  est.

====================================
printer #8 built 9/21/19
flash new mk3s revo hex firmware 3.14
after flashing it automatically started thermal model cal
shows prusa i3 MK3S+R ok on LCD
52453.14m filament, 
total print time 982d

use reset_revo.gcode on sd card
response: file incomplete continue anyway?
yes
Press to reset REVO
Press to Save
0 hours 1 minutes

next, performed PID  calibration to 250

thermal model calibration

====================================
printer #49 built 9/10/23
print meters, days (5781.49m,86 days)
ran reset_revo.gcode
continue anyway? > yes
Press to reset REVO 
press to save

flash new mk3s revo hex firmware 3.14

next, performed PID  calibration to 250
pid cal 0/5 stuck at 31C extruder 26c bed
reset the printer
ran reset_revo.gcode
retry PID cal to 250C

next, thermal model cal

```

Leopere, 2024-07-10:
awesome I haven't been having trouble with mine either on this version but I was waiting because when I had a previous version I didn't have trouble until I did and I don't really know what had changed so I wanted to hear it from someone else before saying anything.

3d-gussner, 2024-07-10:
@mhouse1 Thanks for sharing the information :hugs: . Always nice to see that kind of feedback. :crossed_fingers: The rest of the printers updates are also going that "smoothly"

capsulecorplab, 2024-07-10:
Been running 3.14.0 firmware for the Mk3S with Revo for a couple of days now. Seems to be working so far without errors!

Though, the option to load filament when autoload is turned off seems to have disappeared. Should I open an issue for this?

3d-gussner, 2024-07-11:
@capsulecorplab Thanks for the positive feedback.

The "Load filament" and "Unload filament" menus are shown depending on the state of the filament sensor.
- When filament is detected and printer is idle "Unload filament" is shown and "Load filament" is hidden.
- When NO filament is detected and printer is idle "Load filament" is shown and "Unload filament" is hidden.
- When the FSensor is disabled both "Load filament" and "Unload filament" are shown as the printer doesn't "know" if there is any filament and the user has to be able to use both menus.

Hope that it makes clear.

If you have an question or issue with it, then please open a new Issue to keep this one "clear" to one topic.

mhouse1, 2024-07-11:
So far I  have 4 printers running new fw3.14 successfully 
Two printers are getting thermal anomaly with firmware 3.14 after updating and is unable to perform PID cal. One printer I also tried factory reset to no help. Today, the print farm is running so the room is hot, maybe this is interfering with calibration. I will try this again later when print farm is off. 

```
====================================
printer #10 built 12/16/19
print meters, days (44092.22m,945 days)
ran reset_revo.gcode
file incomplete continue anyway? > yes
Press to reset REVO 
press to save

flashed fw 3.14 thermal model cal
 starts automatically after flashing
fails right away, menu shows Clear TM error above print from sd
cleared TM error

start PID cal to 250C
stuck at nozzle 50/0 bed 28/0 PID cal. 0/5
nozzle is cold
performed power cycle
when i flipped powering switch extruder moves left for some reason
when powering on it tries to resume a print
restart printer using LCD X button
detects thermal model not set
click lcd to continue tries resuming print again, use menu to "Stop Print"
restart printer using LCD X button again
detects thermal model not set, but  does not try to resume print
use reset_revo.gcode
start PID cal to 250C
stuck at nozzle 52/0 bed 28/0 PID cal. 0/5
Perform factory reset all data using instructions at (https://help.prusa3d.com/article/full-system-refresh-original-prusa-i3_133258)
start PID cal to 250C
stuck at nozzle 58/0 bed 28/0 PID cal. 0/5
removed sd card
start calibration > wizard
while extruder is up, realize revo nozzle is a little loose, tightened it
wizard  starts searching bed calibration point
wizard completes and starts thermal model cal
thermal model cal results in thermal anomaly within 1 second
try PID cal to 250 again
PID cal ramps nozzle to 80C then stuck at nozzle ~52/0 bed 28/0 PID cal. 0/5
====================================
printer #40 built 4/26/21
print meters, days (m, days)
ran reset_revo.gcode
file incomplete continue anyway? > yes
Press to reset REVO 
press to save

flashed fw 3.14 thermal model cal
 starts automatically after flashing
next, performed PID  calibration to 250
cannot complete PID cal stuck at PID cal. 0/5
when trying to print from SD LCD Shows PAUSED THERMAL ERROR
restored to fw 3.11 for now so i can use the printer
performed PID cal to 250 and started a print
```

mhouse1, 2024-07-16:
so far 2 out  of 9 printers can't be updated to fw3.14 due to PID cal and thermal model failing right away even though both printers are functional on fw3.11 and is used everyday . Some inconsistencies, some printers starts thermal model cal right away after flashing fw3.14, some doesn't, printer number 10 built 12/16/19 I reverted back to fw3.11 and everything works fine because it kept failing to calibrate on fw3.14 even after factory reset and reset_revo.gcode, when I tried flashing fw3.14 again a few days later it starts self test after flashing. 

for now any printer that accepts fw3.14 I'm updating to that otherwise they get updated to fw3.11 

Details:
```

====================================
printer #10 built 12/16/19 attempting fw upgrade again
has fw 3.11
after flashing 3.14  it starts self test ("First, I will run the selftest to check most common assembly….)
starts thermal cal
fails right away with LCD showing print aborted
run reset revo.gcode
start PID cal, cannot complete PID cal  LCD shows PID cal. 0/5 LCD  menu cannot scroll
reset using LCD button
revert to fw 3.11

====================================
printer #35 built 1/15/21
has fw 3.11
print meters, days (?m, ?days)
after flashing 3.14 it starts thermal cal right away
successful
next, performed PID  calibration to 250
successful

====================================
printer #39 built 4/26/21
has fw 3.11
print meters, days (23681.91m, 386days)

after flashing 3.14 it starts thermal cal right away
next, performed PID  calibration to 250
started small print 33min
successful
====================================
printer #42 built 5/3/21
has fw 3.11
after flashing 3.14 does not auto thermal cal , LCD shows PRUSA i3 MK3S+R Ok.
Start PID cal to 250 … completes successfully
wait for cool down
next, performed thermal model cal
successful
```

capsulecorplab, 2024-07-18:
> Been running 3.14.0 firmware for the Mk3S with Revo for a couple of days now. Seems to be working so far without errors!
> 
> Though, the option to load filament when autoload is turned off seems to have disappeared. Should I open an issue for this?

I spoke too soon. I forgot I had disabled thermal model protection while running v3.13.2 before updating to v3.14.0. Factory resetting the printer brought back the thermal anomaly error until I reran the gcode to disable thermal model protection.

3d-gussner, 2024-07-18:
@capsulecorplab Thanks for the feedback and especially the update comment above.
Please let us know how it is working after the factory reset and TM calibration.

capsulecorplab, 2024-07-18:
> @capsulecorplab Thanks for the feedback. Did the TM cal after the factory reset work?

I tried running TM cal after the reset, but couldn't get the thermal anomaly error to go away, so I ended up just disabling the TM protection

mhouse1, 2024-07-18:
> > @capsulecorplab Thanks for the feedback. Did the TM cal after the factory reset work?
> 
> I tried running TM cal after the reset, but couldn't get the thermal anomaly error to go away, so I ended up just disabling the TM protection

which one is safer, disabling TM protection and using fw3.14 or using fw3.11 ?

mhouse1, 2024-07-18:
no TM issues with fw3.11 but I thought the TM that was added after 3.11 improves on detecting TM anomalies, I wonder if disabling the new TM means we still have what ever safety feature that was in 3.11. Theres definitely lots of improvements in 3.14 that I'd like to use  but if disabling the TM model means its less safe than 3.11 then I won't risk it.

With TM model disabled in 3.14 can it still detect extruder thermal runaway events?  3.11 has been great in detecting thermal runaways which I've see happen 8 times so far in the last 4 years on a print farm of mk3s 50+ printers.

3d-gussner, 2024-08-08:
@mhouse1 Sorry for the late answer.

>With TM model disabled in 3.14 can it still detect extruder thermal runaway events?

Yes the default thermal protection is always active. 
With the TM active, but you might not see it as the TM reacts in seconds while thermal runaway takes longer.

Did you contact E3D support as they should be able to help you with REVO and TM issues,

[33 more replies]