prusa3d/Prusa-Firmware #4453: MMU3 SpoolJoin function application for multi-material prints

Brad-Christy, 2023-10-18:
My problem is that it isn't functional with multi color prints. The dev team needs to work out this function to allow for the assignment of successive spools, so that when a spool runs out, the MMU will subsequently use the assigned successor spool in place of the spool called for in the program. CNC machining centers have been doing this for quite a while now. The machine can be set up so that there are multiple tool pots in the carousel that are assigned to, say, "T1", so that, when the machine decides that the actual "T1" has had it, either by machining time or spindle load, for example, the machine will, from that point, load a tool from a different pot that has been assigned as a successor to T1. This function translates in parallel to a spool running out. I see no reason the SpoolJoin function cannot follow this same behavior. When spool 1 runs out, the MMU uses, say, spool 5 when the program calls for spool 1. Of course this would have to be assigned at the printer, as a follow up to turning SpoolJoin [on]. Simply indicate which spool or spools is/are likely to run out and which spool or spools the MMU is to default to.

3d-gussner, 2023-10-18:
@Brad-Christy Thanks for the idea. Do you mean something like this? https://github.com/prusa3d/Prusa-Firmware/issues/988

Brad-Christy, 2023-10-18:
3d-Gussner,

That sounds like pretty much exactly what I would like to see. I just want it included in the firmware, and not something the average (non-programmer) has to figure out. I can't imagine I'm alone.

In my ideal world, when you go into the settings on the printer, you would turn "SpoolJoin" [ON], then would get prompts that would allow you to assign spools as backups for spools assigned in the GCode. Once assigned, and after a spool has run out, the MMU selects the alternate spool assigned to the initial spool.

Personally, I think the SpoolJoin function should reset to [OFF] after a given print is complete, or at least after the function has been triggered. I would think it beneficial for this to be the case for alternate spool assignments, as well.

I am a little disheartened, though. It would seem that, after five years, if the DevTeam has not implemented something seemingly so appropriate, it has either been determined beyond feasible or unworthy.

As a matter of personal interest, what I am hoping to print on a regular basis entails a large amount of one spool, then the addition of two more spools, with the remainder of the print using all three, as needed, per layer. For this function to be of benefit, the MMU would have to leave spool 1 behind when it was empty, in lieu of an assigned backup, regardless of how many times the GCode called for spool 1.

Thanks. Brad.

3d-gussner, 2023-10-18:
> I am a little disheartened, though. It would seem that, after five years, if the DevTeam has not implemented something seemingly so appropriate, it has either been determined beyond feasible or unworthy.

I am now on the dev team and will ask if that is something we would like to implement. :wink:

Brad-Christy, 2023-10-18:
3d-Gussner,

SaaWEEet!

Thanks. Brad.