prusa3d/Prusa-Firmware #4463: Add octolapse like timelapse to prusa connect and usb camera support

lukylbc, 2023-10-21:
Please add timalepse to prusaconnect but make it like octolapse with each layer it goes of the print and makes the final result much more pleasant to watch. Also im not sure if its possible but please somehow make it possible to use usb cam or rasspberry cam, i personally think that camera in local/cloud printer control is a must, but not everyone has an old reliable phone.

3d-gussner, 2023-10-25:
@lukylbc Thanks for the request but this is the Prusa-Firmware github repository for the MK2.5/S and MK3/S/+ printers
@Prusa-Support Please move this issue to correct repository.

Prusa-Support, 2023-11-11:
We don't have a specific repository for Prusa Connect but most of the GitHub issues about Prusa Connect are gathered in [prusa3d/Prusa-Link-Web](https://github.com/prusa3d/Prusa-Link-Web) as they are usually related to UI/UX.

However, I believe the compatibility with cameras and timelapse features should probably be discussed in [prusa3d/Prusa-Link](https://github.com/prusa3d/Prusa-Link) though, as it seems to be more closely related to the Prusa Link operating system.

Please notice a number of RPi and USB cameras are already compatible, and yet we are working on extending the range of cameras and features. Prusa Link and Prusa Connect are openly available to everyone, but are still in "beta" version.
https://help.prusa3d.com/guide/camera-setup-for-prusalink-prusa-connect_470943

Michele Moramarco
Prusa Research