prusa3d/Prusa-Firmware #4464: [BUG] Revo thermal anomaly during mid-print filament change

fasaxc, 2023-10-22:
**Printer type** -  MK3S with Revo lower power variant
**Printer firmware version** -  3.13.1 Revo version

**MMU upgrade** - none 
**MMU upgrade firmware version** - [e.g. 1.0.6, 1.0.6-RC2, ...]

**SD card or USB/Octoprint**

SD

**Describe the bug**

Not a huge problem but thought it might be useful to report in case it helps tune the thermal model.

Changed filament mid print (manually, just before the filament ran out). When the printer started preheating it hit a thermal anomaly but the filament change went ok apart from that.

I wondered if the thermal model couldn't deal with the preheat step for some reason.
  
**To Reproduce**

- Printing in lack enclosure
- Printing with Prusament galaxy black with Revo 0.8mm.
- Approximately 1hr into print
- open enclosure
- pause print
- select unload filament
- select PLA
- starts preheating and triggered thermal anomaly
  
**Expected behavior**

- unloading filament shouldn't cause a thermal anomaly. Maybe the thermal model should be disabled while changing filament (or made a bit looser)?
  
**G-code**

Can attach the model I was printing if it'd help but I don't have it on my mobile...

**Video**

Sorry, can't easily try again, I only unload filament when necessary because it often jams on unload.

3d-gussner, 2023-10-25:
@fasaxc Thanks for the detailed issue report.
Did you run the Thermal model calibration in the enclosure and for the 0.8mm nozzle?
As we don't have E3D REVO nozzles please contact E3D support. See https://github.com/prusa3d/Prusa-Firmware/issues/4451 

I am closing this issue for the time being, if E3D can't help you please re-open this issue.

closed 2023-10-25 by 3d-gussner

fasaxc, 2023-10-30:
I'm running the revo firmware and I calibrated in the enclosure but with the 0.6mm nozzle IIRC.  Jus tto be clear, the thermal model seems happy during print; it just seemed odd that it instantly triggered when I went to unload filament mid-print.