prusa3d/Prusa-Firmware #4465: [BUG]<Enter comprehensive title>

gysnst2, 2023-10-23:
**Printer type** - [e.g. MK3S, MK3, MK2.5S, MK2.5, MK2S, MK2]
**Printer firmware version** -  [e.g. 3.8.1, 3.8.1-RC1, ...]

**MMU upgrade** - [e.g. MMU2S, MMU2, MMU1]
**MMU upgrade firmware version** - [e.g. 1.0.6, 1.0.6-RC2, ...]

**SD card or USB/Octoprint**
  Please let us know if you print via SD card or USB/Octoprint

**Describe the bug**
  A clear and concise description of what the bug is.
  
**To Reproduce**
  Please describe steps to reproduce the behavior.
  
**Expected behavior**
  A clear and concise description of what you expected to happen.
  
**G-code**
  Please attach a G-code. This will make it easier for us to replicate the error.

**Video**
  Please attach a video. It usually helps to solve the problem.

Flash MK3s to 3.13.1 rc2. Flash mmu2s to 3.0.1

Error mmu firmware need to be upgraded 
[image]

3d-gussner, 2023-10-25:
@gysnst2 Thanks, please add all asked bug information above. We need to know which printer type, printer firmware, etc. to be able to help you.

Where did you find `3.13.1 rc2` ??? See https://github.com/prusa3d/Prusa-Firmware/releases

gysnst2, 2023-10-25:
mk3s+ MMU2S  3.13.2 RC1  and 3.0.1 

3 printers same configuration all stock.

3d-gussner, 2023-10-26:
Did you flash the printer and MMU2S with PrusaSlicer
- the printers with https://github.com/prusa3d/Prusa-Firmware/releases/download/v3.13.2-RC1/prusa3d_fw_MK3S_MK3S+_3_13_2-RC1_7068.hex ?
- and the MMU2S with https://github.com/prusa3d/Prusa-Firmware/releases/download/v3.13.2-RC1/MMU3_3.0.1+837.hex

Please also provide the PrusaSlicer Firmware Update advanced output here.

Please check out that you flashed the MMU2S correctly https://help.prusa3d.com/article/firmware-updating-mmu2s_155118
And please try a MMU2S factory reset https://help.prusa3d.com/article/factory-reset-mmu_510075

If you have a Serial Terminal like Putty see [here](https://help.prusa3d.com/article/crash-dump_364959) please check if the log files shows
```
echo:MMU2:>S0*c6.
echo:MMU2:<S0 A3*22.
echo:MMU2:>S1*ad.
echo:MMU2:<S1 A0*34.
echo:MMU2:>S2*10.
echo:MMU2:<S2 A1*5a.
echo:MMU2:>S3*7b.
echo:MMU2:<S3 A345*75.
```
`S0` get major version number with A3 means 3
`S1` get minor version number with A0 means 0
`S2` get revision number with A1 means 1
`S3` get build number with A345 means 837
So the reported MMU firmware is  3.0.1+837

gysnst2, 2023-10-27:
yes  I used prusa slicer 2.7.0 alpha 1 
i am using 3.0.1 837.hex
and i used the 3.13.2 rc1.

I would like to report that the public release of the mk3s+ firmware 3.13.2 fixed the issue.. on 2 of the printers the 3rd printer is in the middle of a print so i will report back on that one later

gysnst2, 2023-10-28:
ok all 3 of the mk3s+ mmu errors were fixed with the newest 3.13.2 public releases

3d-gussner, 2023-11-06:
@gysnst2 Please consider to close this issue as it has been solved in FW 3.13.2 final.

closed 2024-01-15 by github-actions[bot]