prusa3d/Prusa-Firmware #4468: [BUG]<Enter comprehensive title>

srjacob999, 2023-10-26:
**Printer type** - MK4
**Printer firmware version** -  5.0
**MMU upgrade** - No MMU
**MMU upgrade firmware version** - No MMU

**Does this with both SD card and USB/Octoprint**
  
**Describe the bug**
  Nozzle cleaning fails  See Prusa Forums: https://forum.prusa3d.com/forum/english-forum-original-prusa-i3-mk4-hardware-firmware-and-software-help/nozzle-cleaning-failed-error/
  
**To Reproduce**
  Doing any print, Using PETG, but other filaments do this
  
**Expected behavior**
 Should clean nozzle and pass nozzle cleaning
  
**G-code**
  Please attach a G-code. This will make it easier for us to replicate the error.

**Video**
  Please attach a video. It usually helps to solve the problem.

srjacob999, 2023-10-26:
I don't know how to enter a comprehensive title

3d-gussner, 2023-10-27:
@srjacob999 This is the repository for MK2.5/S MK3/S/+ 8-bit printers. Please open the issue here https://github.com/prusa3d/Prusa-Firmware-Buddy which is form the 32-bit printers MINI, MK4, XL.

closed 2023-10-27 by 3d-gussner