prusa3d/Prusa-Firmware #4469: [FEATURE REQUEST] Missing sheet option in footer interface

gegy1, 2023-10-28:
In last stable firmware version it was possible to set a footer (Settings-> User Interface-> Footer) showing the current sheet on the heat bed. This is missing in current Alpha 2.
This feature was very handy on the mini when you often change the steel plate and do want to check if you have set the right plate before starting a print. Please bring back that feature.

3d-gussner, 2023-10-28:
@gegy1 This is the repository for the 8-bit Prusa-Firmware. Please reopen this issue here https://github.com/prusa3d/Prusa-Firmware-Buddy/issues

closed 2023-10-28 by 3d-gussner

gegy1, 2023-10-28:
Oh, sorry for that. I got here from the alpha firmware post in the forums.
Thank you for the link to the right place.