prusa3d/Prusa-Firmware #4470: [BUG] MMU error screen doesn't trigger "attention" in PrusaLink/Prusa Connect (never sets PAUSED_FOR USER state)

firetech, 2023-10-28:
**Printer type** - MK3S+
**Printer firmware version** -  3.13.2

**MMU upgrade** - MMU3
**MMU upgrade firmware version** - 3.0.1
**SD card or USB/Octoprint**
  Both are affected. However, a connected PrusaLink Pi is needed to actually see the issue. In my case, I'm using a Pi Zero and the RPi port.

**Describe the bug**
  PrusaLink and Prusa Connect don't signal that the printer needs attention when the printer ends up on the MMU error screen. I've tried to navigate the code, and can't find anywhere in the MMU error path (`MMU2::ReportErrorHook()`) that sets the `PAUSED_FOR_USER` state (admittedly, I've only looked quite coarsely), which I think _might_ be the issue. Therefore I'm posting the issue here rather than in PrusaLink/Connect channels.
  
**To Reproduce**
  * Add a Pi with PrusaLink to the printer and connect it to Prusa Connect.
  * Trigger an MMU error during a print.
  * Notice that Prusa Connect still think the printer is printing fine.
  
**Expected behavior**
  Prusa Connect should react to the MMU error and send "Printer requires your attention" notifications.

gudnimg, 2023-10-28:
You’re correct. I remember we decided not to report this status during development (PrusaLink/Connect wasnt really included in the discussion/scope back then) But maybe we can look into it again since the next release is all about supporting PrusaLink/PrusaConnect more… should be easy to add, it just needs to be tested thoroughly.

I can prepare a pull request if you want to try it?

firetech, 2023-10-28:
> I can prepare a pull request if you want to try it?

I could be up for that, yes. I even thought about trying to do it myself, but there were a bit too many question marks hovering above my head to dive into it immediately. :P

gudnimg, 2023-10-28:
@firetech I haven't tested the changes in the pull request above, but I'm pretty sure it will work. I can test it tomorrow.

closed 2023-10-31 by 3d-gussner

gudnimg, 2023-10-31:
@firetech thank you for opening this issue. We’ve merged the changes :)