prusa3d/Prusa-Firmware #4475: [BUG] Filament Sensor ON/OFF-menu does not work

krite, 2023-11-01:
MK3
3.12.x -> current
No MMU
SD Card

Noticing the error during ABS-printing, it's described in issue #3872, the printer randomly starts a filament-change procedure.
Previously, you could turn off the filament sensor to ignore this kind of issue, but in release 3.12.x and forward, the menu-option for this does not work any longer.
The menu shows the FS as off, but it is really on. Trying to change the status the submenu shows 'OFF' as well, but going back upp after clicking the menu-option, nothing happens.
Downgrading to 3.11.0 makes the menu work again, showing correct status and being able to switch the FS on and off correctly.
  
**To Reproduce**
Try to turn the FS on and off, it clearly does not change.
  
**Expected behavior**
A working menu and option to turn the FS on and off.
  
**G-code**
Not relevant.

gudnimg, 2023-11-03:
Hi, I can try look into this tomorrow, but it would be great to get some info confirmed:
- You specify "MK3", is it a stock MK3 or have you upgraded the filament sensor anytime?
- Can you confirm the issue is also present in 3.13.0, 3.13.1 or 3.13.2?

gudnimg, 2023-11-03:
If you turn on the filament sensor, do the runout, autoload, jam menus show "N/A"? Or do they show correctly ON/OFF?

On 3.13 "N/A" means there is an error, I wonder if that is the case here.

3d-gussner, 2023-11-29:
Closing as no response in three weeks.

krite, 2023-12-21:
Sorry, have been offline for a while.

You specify "MK3", is it a stock MK3 or have you upgraded the filament sensor anytime?
- It is a stock MK3, no FS-upgrades.

Can you confirm the issue is also present in 3.13.0, 3.13.1 or 3.13.2?
- Yes the bug is there in all later versions I tested.

 If you turn on the filament sensor, do the runout, autoload, jam menus show "N/A"? Or do they show correctly ON/OFF?
- The sensor cannot be turned off or on on these firmwares, the menu does not work.

On 3.13 "N/A" means there is an error, I wonder if that is the case here.
- Moving back to the 3.11-fw makes the sensor, menus and options work as before again.

To me, it sounds as a pure software/firmware-related bug in 3.12 and up.

To not have the FS active is not a big deal I can argue (always print with full spools), but having the printer stop printing randomly is!

I'm now stuck at 3.11, so, yeah, please don't close this issue before investigating it further.

3d-gussner, 2023-12-21:
Sorry but I can't reproduce your issue.

https://github.com/prusa3d/Prusa-Firmware/assets/25530011/92ebeffa-739a-4bf1-b090-c543b4b38146

krite, 2023-12-21:
I see, this will make this issue harder to fix of course.
I tried both MK3_3_13_1_6876 and MK3_3_13_2_7080 yielding the same issue. Reverting to MK3_3_11_0_4955 fixes it.
Tried some 12-version in-between as well, which is where the issue actually started occurring.
So, is there any code-changes between those releases that could cause this behaviour?

closed 2024-02-28 by github-actions[bot]