prusa3d/Prusa-Firmware #4476: [BUG] 3.13.2 Layer Shifting MK3S

jorgelserve, 2023-11-01:
**Printer type** - [MK3S, MK3S]
**Printer firmware version** -  [3.13.2, 3.13.2]

**SD card or USB/Octoprint**
Printing via *SD card*

**Describe the bug**
I have four Prusa MK3S printers. All four printers were running [3.13.2-RC1](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.13.2-RC1) and were printing normally. When I updated the firmware on all four printers to [3.13.2](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.13.2), the two printers started to lose steps on the Y-axis. Also tried [3.13.1](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.13.1) and works fine
  
**To Reproduce**
1. Update four Prusa MK3S printers to Prusa-Firmware v3.13.2.
2. Slice your g-code using PrusaSlicer v2.6.1
3. Print any model in **normal** mode that requires significant Y-axis movement.

**Expected behavior**
The Y-axis motor should not lose steps and the print should not show layer shifting.

**Video**

https://github.com/prusa3d/Prusa-Firmware/assets/9343755/c529e226-13f4-4fe1-ad87-eec2a6010b4e

firetech, 2023-11-03:
Not sure if this is relevant, but I'm having weird issues with my **Z** axis on 3.13.2 on my MK3S+. I think I've ruled out basically everything but the firmware at this point, and this ticket made me think there might be (rare?) motor control issues on 3.13.2.

My printer prints fine, but some amount of prints (~10%?) end with the Z motors fighting each other during the final lift:

https://github.com/prusa3d/Prusa-Firmware/assets/104522/b874a0dc-b3d9-4e19-a2bd-f0d6225ff3ff

I've only ever seen this happen during that final lift, but I've also recently had some weird issues in prints that involve big Z axis movements (like one multi material print with "No sparse layers", the result of which can be seen towards the back of the print sheet in the video, and one "Complete individual objects" print). Otherwise prints work fine. Both Z calibration and moving Z with the Move Axis menu also work without issues. The first time this happened was just after upgrading to 3.13.2, and reverting to 3.13.1 has (so far) made the problem disappear.

If this isn't relevant here, I can open a separate ticket instead.

BallieBall, 2023-11-08:
> Not sure if this is relevant, but I'm having weird issues with my **Z** axis on 3.13.2 on my MK3S+. I think I've ruled out basically everything but the firmware at this point, and this ticket made me think there might be (rare?) motor control issues on 3.13.2.
> 
> My printer prints fine, but some amount of prints (~10%?) end with the Z motors fighting each other during the final lift:
> Z_growl.mp4
> 
> I've only ever seen this happen during that final lift, but I've also recently had some weird issues in prints that involve big Z axis movements (like one multi material print with "No sparse layers", the result of which can be seen towards the back of the print sheet in the video, and one "Complete individual objects" print). Otherwise prints work fine. Both Z calibration and moving Z with the Move Axis menu also work without issues. The first time this happened was just after upgrading to 3.13.2, and reverting to 3.13.1 has (so far) made the problem disappear.
> 
> If this isn't relevant here, I can open a separate ticket instead.

I had the same issue on my mk4. My very first print i printed a simple pla fllament template and randomly my nozzle ran away from the print and it scared me tbh. I ignored it and a few prints later everything was fine. I then started a large petg 5 hour print and 2 hours in i hear loud vibrations and my layer shifted very wide. Belts all seems completely fine, nothing seems abnormal so i had no choice but to remove the alpha firmware to see if anything odd happens again, which is where i am now. Currently doing the same 5 hour petg print. hopefully no problems.

zepyrshut, 2023-11-15:
I what I have is yesterday's commit build of the firmware and I have the same problem. When it takes the print about 10-15 layers, it stops printing and the head goes to the starting point.

It is very strange, in the afternoon I will compile an earlier stable release and try.

Roonster78, 2023-11-22:
I've got exactly the same behavior with both of my MK3S+ running 3.13.2. If I switch back/downgrade to 3.12.2 everything works fine and without any issues. Tried it several times. Both printers are completely stock.

Housey5450, 2023-11-22:
Same behaviour here. On 3.13.2 my x and y axis can easily be moved by hand when printing. The weak motors are causeing layer shifting. When I downgrade to 3.13.0 I can slide the printer on my desk holding the extruder whilst printing without causeing the motors to skip. There is a massive difference in motor strength between the two. I have tried this with and without MMU2S installed, same result.

wxves, 2023-11-26:
I have three Prusa MK3S's as part of my print farm. Purchased them around half a year ago. After updating all three of them to firmware 3.13.2-7080 they started having problems like the ones mentioned here. First test with one of the printers resulted in a failed print because of a layer shift. Second and third print with the same machine showed weird overextrusion around the same layer. Printer number two only showed a slightly 'thicker' more prominent layer, apart from that it printed just fine like before. Printer number three showed zero issues at all strangely enough. 

After slicing the exact same stl that previously failed and creating a new gcode file to test, the overextruded layer didn't go away but instead moved up multiple layers. It also still showed ringing. 

I'll downgrade the two malfunctioning printers to firmware 3.12.2 like @Roonster78 mentioned to see if the issues I'm having going away.

[image]
[image]
[image]
[image]
[image]

gudnimg, 2023-11-26:
For the "thicker" layers or layer shifts.... do the LCD statistics show any motor crashes were detected? The firmware should re-home the printer so I wouldn't think that would create a shift but I still wonder 🤔

If you have crash detection disabled, that would also be good information to have.

wxves, 2023-11-26:
All printers run stealth mode so crash detection is disabled. Currently doing a test print on the 3.12.2 firmware and so far so good. 🤞🏻

wxves, 2023-11-27:
After downgrading both two printers to 3.12.2, they printed the exact same gcode file flawlessly. Something is clearly wrong with the 3.13.2 firmware version!

Roonster78, 2023-12-18:
Hm, still waiting for a new beta with this bug fixed but nope. I guess they have all hands on deck to fix the bugs in the Buddy firmware.
I've started a 15h print yesterday with nice PC-CF and totally forgot that I still had 3.13.2 on this printer. Well, I've got reminded this morning - several layer shifts, print ruined, time and material wasted. So I did a downgrade to 3.12.2 and it is printing the same gcode without any shifts or other issues.

jorgelserve, 2023-12-21:
**Update**
All printers are now experiencing this issue with the [3.13.2](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.13.2
) firmware version. I had to revert all printers back to the [3.13.2-RC1](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.13.2-RC1) version to resolve the issue. This is a significant inconvenience as it disrupts our printing workflow and affects the quality of our prints. I'm attaching a photo of multiple failed parts due to step loss. It's quite frustrating that a bug of this magnitude has not been addressed yet. I hope this issue can be prioritized for a fix in the next firmware update.
 [image]

3d-gussner, 2023-12-22:
@jorgelserve Thanks for the update. So FW 3.13.2-RC1 works correctly and FW 3.13.2 final not, right?
Here the changes between these two versions https://github.com/prusa3d/Prusa-Firmware/compare/v3.13.2-RC1...v3.13.2
@gudnimg I assigned you to this issue as you commented here already

firetech, 2023-12-22:
Just to add a datapoint: My (possibly irrelevant) Z axis issue (see first comment of this ticket) was not solved by downgrading to 3.13.2-RC1, I had to go back to 3.13.1 to get rid of that issue.

3d-gussner, 2023-12-22:
@firetech Thanks to the update...

firetech, 2023-12-22:
> changes from 3.13.1 to 3.13.2 are bit bigger than above

I am aware. Since I currently run a custom 3.13.1 with some fixes backported from 3.13.2, I can at least exclude the "temperature jump to 200/50°C when setting temperature manually" feature and the TEMP_HYSTERESIS fix from the list of possible culprits. 😅 (Not that they were likely culprits to start with...)

If the issue was less random than it is, I'd spun up `git bisect` to try to find it, but I'm having issues reproducing it even when I know it's there...

jorgelserve, 2023-12-24:
> @jorgelserve Thanks for the update. So FW 3.13.2-RC1 works correctly and FW 3.13.2 final not, right?

@3d-gussner Downgrading to 3.13.2-RC1 resolved the step loss issue. However, other two printers exhibited X-axis step loss before the downgrade. A coworker also reported observing Y-axis step loss on one of these printers

cybergogo, 2023-12-31:
Hello all,
I also have the same issue which create random nasty lines on the Z axis. 
Also, the MMU2S has trouble pulling the filament at the end of a print, like if the extruder was acting as a brake. Have to pull it manually. 😪
Pretty crazy to have those issues on a final firmware...
[image]
[image]
[image]

Prusa-Support, 2024-01-25:
This is still being investigated but apparently hardly reproducible.

I wonder if all the reporters in this issue attempted a **factory reset**, or tried comparing the results between normal power mode and stealth power mode.

I noticed that some of you tried to move the axis while printing, encountering lower resistance, but that should not be tested while the printer is moving. You can try that while the X and Y motors are engaged, possibly in static conditions (non-printing).
1) Run autohome.
2) Move the X, Y, and Z axes somewhere in the middle of the axis, via "settings > move axis". This way the motors will be engaged. 
3) Try to slide the X or Y axis by hand without the use of force.
4) Repeat the test in _Normal_ and _Stealth_ power mode on FW 3.13.2
4) Downgrade to FW 3.13.1 or 3.13.0 and repeat to compare the motor resistance opposing your hand.

Yet another way to see if the amount of friction varies between firmware versions could be via the belt test.
1. First please clean the X and Y smooth rods with a tissue that doesn't leave residues, and then lubricate the rods using one of the [approved lubricants](https://help.prusa3d.com/article/i3-printer-regular-maintenance_2072). Slide the axis from end to end a few times and remove all the visible lubricant leftovers. This should improve the belt test accuracy.
2. Follow "calibration > belt test" and write down the reported values.
3. Downgrade to FW 3.13.1 or 3.13.0 and repeat from point (1) for better repeatability of the test, and compare the values at the end of the belt test.

Michele Moramarco
Prusa Research

Purplegopher, 2024-02-11:
Hi there. I had the same issue on 3.12.2 and it went away after reverting back to 3.12.1. Found this report afterwards.

I can't attach the gcode, but it would always have those "fat layers" at random heights. I was pulling my hair out. I had upgraded my firmware after not using my MK2s+ for several months.

I had tried printing from Octoprint or SD card, no difference.
I had varied the speed with 100%, 120% (that's where I only got "two" bulging lines, but could be just randomly lucky), 125%, 200% directly on the printer.
All those with the same gcode file.
Printer was always in stealth mode.
Sliced with 2.7.1 on linux, unmodified 0.3draft profile for prusament pla.

I tried printing a simple cuboid taller than the real model and the problem did not happen.

After downgrading to 3.12.1, the same gcode file printed fine again.

(Image taken from defective prints on 3.13.2, picture shows them laying flat on one side. The two on the left were aborted after a few layers. All surfaces are smooth in the print with 3.13.1)
[image]

cybergogo, 2024-02-14:
Hello Michele,

this is not against you personally but I find this answer to be totally unappropriated.
I own 5 of your machines, each one with a MMU2S.
The MMU2S has been a pain since release, but I didn't have much of a choice to do multicolor printing until now.
I've been printing and testing multiple corrections to overcome a badly engineered module.
Those new bugs have made me throw away multiple 24h prints, which could have been avoided.

Now I see that I should be one of your test dummies for you guys. No way.
My machine are in mint condition and fully maintained.
Those tools are my work and I can't do Prusa's work that should be testing the issue yourself.
I only have 5 printers and I can see those issue propagate.
I can't believe that you guys can't reproduce it.

Instead of releasing new products you should take care of your core customers and take more time to test your firmware before release.

So let me know if you have some news for us, your customers.
Until then, I'm looking at Bambu lab's A1 to replace the unreliable machines I have at the moment.

Will be waiting for your reply.

  5.  Downgrade to FW 3.13.1 or 3.13.0 and repeat to compare the motor resistance opposing your hand.

Roonster78, 2024-02-14:
**Hi, my response(s) in bold**
> This is still being investigated but apparently hardly reproducible.
> 
> I wonder if all the reporters in this issue attempted a **factory reset**, or tried comparing the results between normal power mode and stealth power mode.
>  **YES, I did it on my two MK3S+ several times and it is every time PITA and time consuming to get them back calibrated and working again. But they don't work properly with any 3.1.3.x release, I have always to go back to 3.12.2-5713 to have fully working printers, regardless stealth mode or not.**
> I noticed that some of you tried to move the axis while printing, encountering lower resistance, but that should not be tested while the printer is moving. You can try that while the X and Y motors are engaged, possibly in static conditions (non-printing).
> 

**I'm sorry but the following steps you have listed here look like the standard answers from customer support.
If I switch back and forth between the firmware and change nothing, absolutely nothing on the hardware and the error only occurs with firmware 3..13.x but not with 3.12.2-5713, it is a software problem and not a hardware problem.
All my Prusa printers are set up perfectly and are serviced regularly. The problems only occurred with firmware 3.13.x.**

> 1. Run autohome.
> 2. Move the X, Y, and Z axes somewhere in the middle of the axis, via "settings > move axis". This way the motors will be engaged.
> 3. Try to slide the X or Y axis by hand without [...]

**I gave every 3.13.x release a chance, tested it and then had to go back to 3.12.2. This means I lose time and money unnecessarily.**

Segaertj, 2024-02-15:
Having the same issues with 3.13.2 after rebuilding my printers now. One of my machines has it, the other one hasn't. Trying a factory reset and calibration again, if that doesn't work, I'm reverting to 3.11.0 (the first downgrade version with support for thermal calibration with a Revo if I'm correct?)

Pictures (black parts suffer from it, grey ones are alright):
[image]

Prusa-Support, 2024-02-20:
No worries, I think our developers collected sufficient information from other sources. Thanks for your answer anyway.
Our developers are currently preparing a fix for a possibly related bug.

.

@Purplegopher That seems to be a known phenomenon AKA the "[hull line](https://help.prusa3d.com/article/the-benchy-hull-line_124745)" on the benchy boat, but probably mixed with low temperatures. I may be wrong but our [Customer Support](https://help.prusa3d.com/article/customer-support_2287) can evaluate if this is the case and help.

Michele Moramarco
Prusa Research

Roonster78, 2024-03-15:
Hello everyone, hello Michele,
After careful testing and more than 10 prints per machine, I can confirm, at least for myself, that the problems with firmware 3.13.3 have been resolved.

- XYZ calibration works without problems
- no layer shifts on X & Y axis
- no problems with the Z-axis

Best regards
Roonster

3d-gussner, 2024-03-18:
@Roonster78 Thanks for the feedback and testing, very appreciated.
 
We hoped for some more feedback from everyone here reporting. 

But as we didn't get any other feedback, I will close this issue.

If anyone has still an issue feel free to reopen or commit here.

closed 2024-03-18 by 3d-gussner