prusa3d/Prusa-Firmware #4478: Extruder motor whistles, vibrates and axis does not rotate in version 3.13.x during first layer calibration

Juliendu78, 2023-11-03:
Extruder stepper motor malfunctions as soon as first layer calibration begins
No problem with Move Axis > Extruder, the extruder motor runs perfectly and the filament flows correctly.

On the other hand, if I run a first layer calibration, after having done the mesh bed leveling. As soon as it tries to extrude filament, the stepper motor makes a strange whistling sound. The motor shaft doesn't turn. The motor vibrates and whistles, as if it were turning inside, something is happening, but the motor shaft remains stationary.

If I let the first-layer calibration take place, the motor shaft starts to turn sporadically after a while.

No problem with version 3.12.2. I have this bug with firmware 3.13.x. I've changed firmware several times, and it's always the same.

At first, I thought it was a motherboard fault that had suddenly appeared, linked to the stepper motor driver. Or a defect in the stepper motor (but I replaced it with another, leaving the motor running in free air, without engaging anything, and the behavior was the same).

Really a firmware problem / A specific hardware problem linked to my motherboard in particular?

3d-gussner, 2023-11-12:
@Juliendu78 Please add the template asked information https://github.com/prusa3d/Prusa-Firmware/issues/new?assignees=&labels=bug&projects=&template=bug_report.md&title=%5BBUG%5D%3CEnter+comprehensive+title%3E

How do you make FW 3.13.x firmware? Do you compile it yourself?

Juliendu78, 2023-11-24:
We can close the bug, It was stock pre-built firmware. In the end, I did a factory reset. And now the printer is working properly. Thanks  @3d-gussner

3d-gussner, 2023-11-24:
@Juliendu78 Thanks for letting us know. Please close this issue yourself looks always nicer when the owner of the ticket does that instead of us.

jorgelserve, 2024-01-25:
@Juliendu78 downgrade to [v3.13.2-RC1](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.13.2-RC1) fix the steeper motors the issues for me (#4476)

3d-gussner, 2024-03-13:
Closing as no interaction

closed 2024-03-13 by 3d-gussner