prusa3d/Prusa-Firmware #4480: [BUG] 3.13.2 With REVO the nozzle the temperature is wrong (not the real one)!

antimix, 2023-11-05:
**Printer type** - MK3S+
**Printer firmware version** -  3.13.2 REVO
**MMU upgrade** - MMU3
**SD card FlashAir**

**Describe the bug**
The real nozzle temperature is around 5° less than the LCD reported. This affect prints with low temp materials as PLA.
  
**To Reproduce**
Fit a temperature sensor probe inside the extruder and reach the REVO nozzle till the bottom. Read how the read values are different than the values on the LCD.
  
**Expected behaviour**
Real temperature reported +/- 0.5%
  
 **Notes**
 The Temperature Model setup was just done.
 
[image]
[image]

3d-gussner, 2023-11-06:
@antimix Thanks for the report, but please contact E3D.
They should be able to provide more information how the thermistor works.
As far I can see the REVO uses the same thermistor number 5 in the table as Prusa stock E3D v6 thermistor.

sarusani, 2023-11-06:
The thermistor reads the temperature at the heat block not in the nozzle. There will always be some difference, with any hotend. The material of the nozzle will also influence the temp.

This is pretty normal with every hotend I ever owned. Just adjust the temp in the slicer.

jamesdehart, 2023-11-30:
@antimix - Did you get any response from your ticket with E3D?

antimix, 2024-02-01:
What I could add is that I have noticed that when I start "Preheat" the printer start heating, when the nozzle temperature reach around 50°C it raise always THERMAL ANOMALY on the LCD. The message do not disappear. 
- is it a bug ?

**Example:** 
from cold, go on the LCD and select Setting->Temperature->Nozzle: set to 50. 
It will reach the 50°, but just around that will write THERMAL ANOMALY on the bottom of the LCD. The temperature will beached and it will be stable.
Now move to 60° and again the nozzle will reach 60° steady degree, but the THERMAL ANOMALY message will always stay on the bottom of the LCD.

- Is there any way to clear the error message ?
- Is really an error message or is just a bug that writes it ?

3d-gussner, 2024-03-08:
@antimix Please read https://github.com/prusa3d/Prusa-Firmware/issues/4451 and follow the instructions. Also E3D should be able to help you with this issue.

>  Is there any way to clear the error message ?

Normally the message will be cleared by the next Status Line update like a new message or when a filename is shown

> Is really an error message or is just a bug that writes it ?

The message is shown when a thermal model warning occurs
But also when a thermal model error is reported. This can be then cleared in the Tune menu during a print.

3d-gussner, 2024-04-02:
Closing due to no interaction.

closed 2024-04-02 by 3d-gussner