prusa3d/Prusa-Firmware #4482: [BUG] FW 3.13.2 MK3s - no option to load filament on pause print

PlinyElder, 2023-11-06:
**Printer type**  MK3S+
**Printer firmware version** -  3.13.2

**MMU upgrade** - N/A
**MMU upgrade firmware version** -N/A 
**SD card or USB/Octoprint** - SD card
**Describe the bug**
During a print, when the printer pauses, there used to be an option to unload and load filament. That option seems to be missing when the printer pauses.
  
**To Reproduce**
  Print a part that has a pause for a filament change 
  
**Expected behavior**
  The print should pause and present the option for unloading filament, then resuming the print
  
**G-code**
  Please attach a G-code. This will make it easier for us to replicate the error.

**Video**
  Please attach a video. It usually helps to solve the problem.

dzwiedziu-nkg, 2023-11-19:
Do you try turn off from power for few minutes and turn on? My filament sensor (v0.4) hangs after upgrade from 3.13.1 to 3.13.2. Still not working after downgrade to 3.13.1. But when I turn off from power and turn on, filament sensor works correctly and still work after upgrade to 3.13.2.

Zeuster56, 2023-12-12:
I have the exact same issue as PlinyElder, same setup as well. Sometimes the unload option is there and the next time it isn't? I went from 3.12.1 to 3.13.2. I only use the SD card option.

Printer type MK3S+
Printer firmware version - 3.13.2

MMU upgrade - N/A
MMU upgrade firmware version -N/A
SD card or USB/Octoprint - SD card
Describe the bug
During a print, when the printer pauses, there used to be an option to unload and load filament. That option seems to be missing when the printer pauses.

To Reproduce
Print a part that has a pause for a filament change

Expected behavior
The print should pause and present the option for unloading filament, then resuming the print

Zeuster56, 2023-12-12:
Thank you so much!!!

bjanders, 2024-02-04:
The Change filament under Tune does not work while in Pause. The idler engages neither for unloading nor loading.

I noticed the filament was running out during a print, so I paused it to change it. I found Change filament under Tune. The "pull out filament" message was shown, but it did not push it out with the idler. I pulled it forcefully thinking it was some temporary issue. Then I inserted the new filament. The idler did not pull it in, but showed the "was the filament inserted correctly" message. I answered "no" and the same happened again (the idler did not  engage). I then just selected "Yes" without any filament and resumed the print, thinking it would notice there was no filament and ask me to insert, but it just started printing without filament. I then selected without pausing "Change filament" from the Tune menu, and it worked as expected.

(As a side note, I don't find it intuitive that that Change Filament is under Tune, as it is not a tuning operation, unlike, say, Z height tuning).

3d-gussner, 2024-03-08:
@bjanders When you run out of filament it will start the change procedure.
When you pause the print in FW 3.13.3 and lower you gonna need to go to Tune and set the the nozzle temperature as it is turned off during pause. Then you can select change filament from tune menu.

As the hotend is turned off during pause, to prevent filament cooking, you may have experienced that the hotend was too cold to pull the filament out and push the new in.    

This has been fixed and will be in FW 3.14.0

> (As a side note, I don't find it intuitive that Change Filament is under Tune, as it is not a tuning operation, unlike, say, Z height tuning).

Changes to the main menu aren't easy to get approved as changes to sub menus.  

Please consider to close this issue.

bjanders, 2024-03-08:
> @bjanders When you run out of filament it will start the change procedure. 

Yes, I know, but sometimes it gets stuck trying to eject the bit left, since typically the last end it slightly bent where it has been attached to the spool. So if I notice the filament is running out I do a manual change just before all filament has disappeared in the head.

> When you pause the print in FW 3.13.3 and lower you gonna need to go to Tune and set the the nozzle temperature as it is turned off during pause. Then you can select change filament from tune menu. [...]

But why did it show "Pull out filament" if the nozzle was too cold?  How would I be able to heat the nozzle, if the "pull out filament" message is shown, i.e. how do I access the menu?  Same thing when inserting, it said "insert filament", but then didn't do anything when inserting. Normally it would automatically start heating the head if it's too cold when it senses the filament. Or did I just not notice it started heating, and did not wait long enough?

> This has been fixed and will be in FW 3.14.0

So there was bug related to this, or is this fix referring to something else in this thread? Which commit is this referring to?

> > (As a side note, I don't find it intuitive that Change Filament is under Tune, as it is not a tuning operation, unlike, say, Z height tuning).
> 
> Changes to the main menu aren't easy to get approved as changes to sub menus.

When not printing, the unload filament is not under Tune, so it's confusing that location changes depending on whether you are printing or not.

> Please consider to close this issue.

I assume this addresses the original reporter, and not me?

3d-gussner, 2024-03-08:
@bjanders The issue has been fixed with https://github.com/prusa3d/Prusa-Firmware/pull/4408 restoring the temperature.