prusa3d/Prusa-Firmware #4484: [BUG] Setting MMU Bowden length doesn't work

mdjurfeldt, 2023-11-06:
**Printer type**

MK3S+

**Printer firmware version** 

3.13.2

**MMU upgrade**

MMU3

**MMU upgrade firmware version** - [e.g. 1.0.6, 1.0.6-RC2, ...]

3.0.1

**Describe the bug**

I've tried to set Bowden length using

`M708 A0x22 X472`

but when I read it back using M707, I get the default value (0x168 = 360).

A hint of what is going on might be seen (for an expert, not me) in the log:

```
Send: M708 A0X22 X472
Recv: echo:MMU2:>W22 16*22.
Recv: echo:MMU2:<W22 A*1.
Recv: ResetRetryAttempts
Recv: ok
```

3d-gussner, 2023-11-06:
@mdjurfeldt The `M707/M708` are case sensitive and you need a serial terminal that doesn't UPPER CASE automatically. Which serial terminal did you use?

Pronterface UPPER CASE automatically so please don't use it!
In OctoPrint you need to add `M707` and `M708` to **_Settings -> Features -> Terminal Auto Uppercase Blacklist_**

Send `M708 A0x22 X472` with a lower case `x` for the register `A`dress `22`

mdjurfeldt, 2023-11-06:
Thanks, that worked. Well, I've seen a lot of people having similar problems with MMU registers. Wouldn't it be a good idea to accept 0X22 as a valid register name? Closing this issue...

closed 2023-11-06 by mdjurfeldt

3d-gussner, 2023-11-06:
> Wouldn't it be a good idea to accept 0X22 as a valid register name? Closing this issue...

It would be but our Gocde parser is very limited and the `X` is used to set the value. We have limited resources on this 8-bit ATmega.
Also this `G-code M708` and few others aren't used that often as most users stay with the stock configuration.

Can you share a picture of your setup? Always interesting how the community use the MMU and buffers.

mdjurfeldt, 2023-11-06:
@3d-gussner Sure. I only use my buffer when I have a lot of colors in the model, and now I'm considering trying out your new one, which apparently was inspired by one of my makes (I was acknowledged in the release announcement for MMU3). For the usual soluble support + single color model, I use the spannerhands boxes, one of which you see on top of my enclosure:

[image]

The purpose of the enclosure is to get a hot environment for some filaments as well as filtering away particles and toxic fumes. The Bowden tube runs through a bearing which sits in a 3D-printed flexible structure which easily follows the MMU selector movements:

[image]
[image]

I've planned to construct a mechanism which lifts the Bowden tube out of the enclosure with some slight force in order to get the filament path more straight with less resistance as well as avoiding strange Z-forces.

I'm pretty happy with the way I've arranged the cabling. It's easy to unplug the printer from the enclosure. It's just a couple of XT30 plugs as well as the Molex C-grid connector for the MMU communication. When unplugged I can use the PSU still attached to the printer and it's pretty easy to re-attach the MMU on the frame.