prusa3d/Prusa-Firmware #4486: [BUG] FW 3.13.2 MK2.5S - Filament sensor not working

dwprod38, 2023-11-08:
**MMU2S** - firmware version 3.0.1

**To Reproduce**

After flashing MMU2S to 3.0.1 and MK2.5S to 3.13.2 (prusa3d_fw_MMU3_3_0_1.hex and prusa3d_fw_MK25S_3_13_2_7080_RAMBo13a_en-fr.hex) with prusaslicer 2.6.1 --> Warning message: "printer type changed". After clicking button, second message appeared "Old settings found. default PID, Esteps etc. Will be set." And I have filament sensor didn't working - 

In system menu/Sensor info : FS bit and FINDA didn't change their value when pushing filament and PiNDA works fine alone.

In Setting/Fil. Sensor menu, we have:
Fil. Sensor [ON]
F. Runout [N/A]
F. Autoload [N/A]
FS Action [Cont.]

Back flashing MMU2S to 3.0.0 and MK2.5S to 3.13.1 (prusa3d_fw_MMU3_3_0_0.hex and prusa3d_fw_MK25S_3_13_1_6876_RAMBo13a_en-fr.hex) with prusaslicer 2.6.1 --> Warning message: "printer type changed". After clicking button, second message appeared "Old settings found. default PID, Esteps etc. Will be set." 

Filament sensor works fine - In system menu/Sensor info : FS bit and Finda MMU2S change value when pushing filament, PiNDA works fine too.

In Setting/Fil. Sensor menu, we have:
Fil. Sensor [ON]
F. Runout [ON]
F. Autoload [ON]
FS Action [Cont.]

After second attempt flashing MMU2S to 3.0.1 and MK2.5S to 3.13.2 (prusa3d_fw_MMU3_3_0_1.hex and prusa3d_fw_MK25S_3_13_2_7080_RAMBo13a_en-fr.hex) with prusaslicer 2.6.1 --> Warning message: "printer type changed". After clicking button, second message appeared "Old settings found. default PID, Esteps etc. Will be set." 

Filament sensor didn't working - In system menu/Sensor info : FS bit and FINDA didn't change their value when pushing filament and PiNDA works fine alone.

In Setting/Fil. Sensor menu, we have:
Fil. Sensor [ON]
F. Runout [N/A]
F. Autoload [N/A]
FS Action [Cont.]

Files links used:

https://cdn.prusa3d.com/downloads/firmware/prusa3d_fw_3_13_2_MK25S.zip

https://cdn.prusa3d.com/downloads/firmware/prusa3d_fw_3_13_2_MK3S_3_0_1_MMU2S.zip

https://cdn.prusa3d.com/downloads/firmware/prusa3d_fw_3_13_1_MK25S.zip

https://cdn.prusa3d.com/downloads/firmware/prusa3d_fw_3_13_1_MK3S_3_0_0_MMU2S.zip

gudnimg, 2023-11-08:
“printer type changed”: you should never get this messge unless you flashed firmware which is for a different printer (For example MK3, MK3S)

The LCD message should say exactly what the type changed to and what the type was before. One of them is wrong.

dwprod38, 2023-11-08:
Yes I understand that used firmware file is directly published from the Prusa3D site : https://cdn.prusa3d.com/downloads/firmware/prusa3d_fw_3_13_2_MK25.zip

And after unzipping it : there's only hex files for Rambo10a and Rambo13a. None files dedicated to Einsi board used by MK3 were found in this zip... So something wrong with labelling/compiling hex files by Prusa developers...

gudnimg, 2023-11-08:
If you have MK3 then you should use this: https://cdn.prusa3d.com/downloads/firmware/prusa3d_fw_3_13_2_MK3.zip

closed 2023-12-31 by dwprod38