prusa3d/Prusa-Firmware #4488: Moving nozzle size choice to the first page of the menu.

LoakAudio, 2023-11-10:
**Printer type** - MK3S

Hi devs,

I'm one of the 80% guys who doesn't change often the nozzle on E3D v6 hothends ... because I'm lazy.
With the arrival of the Revo, laziness is not a reason anymore.
But when you are lazy ...
So, could it be possible to move the nozzle size item of the the menu to the first page of the menu, let's say under the sheet choice ?

LoakAudio, 2024-01-09:
I've just seen that this feature request was taken into account.
If you need volunteer for test feel free to ping me, if you need volunteers for code, it's far out of my skill.

I don't know if it is the place to talk about it, but in the Revo firmware, the part concerning the preheat of the hotend to remove the nozzle could be removed from the firmware to free up some space.

3d-gussner, 2024-01-10:
@LoakAudio Thanks for volunteering to test this. The implementation isn't that hard but as the LCD menu changes it needs some consideration. There are plenty of help articles and documents about the LCD menu layout and you can imagine that changing every related article and documentation isn't that easy.
On the other hand this additional menu only makes sense with a quick changeable nozzle like the REVO.
The REVO builds are "community made" builds so changes to the articles and documents may not be needed.

3d-gussner, 2024-01-10:
@LoakAudio Please have a look at #4563 Is this what you had in mind?

LoakAudio, 2024-01-10:
Thanks ! That's exactly what I had in mind with a bonus with main screen.
What will be the main screen during a print ? I mean nozzle size won't take the place of remaining time ?

3d-gussner, 2024-01-10:
>What will be the main screen during a print ? I mean nozzle size won't take the place of remaining time ?

While printing the Info screen shows the normal screen with Type of print SD/HO + percent and the Estimated Time
See https://github.com/prusa3d/Prusa-Firmware/pull/4563/files#diff-5fbfdca43c192573e31d8e78d255c510fb6e213b118cf38173277ec2a2163880R640

I am not sure if this PR will be approved even it is only for the community REVO builds. Some :+1:  from the REVO community would be helpful.

3d-gussner, 2024-04-03:
Closing as https://github.com/prusa3d/Prusa-Firmware/pull/4563 has been merged

closed 2024-04-03 by 3d-gussner