prusa3d/Prusa-Firmware #4489: [BUG]<Enter comprehensive title>

ruedli, 2023-11-10:
**Printer type** - [e.g. MK3S, MK3, MK2.5S, MK2.5, MK2S, MK2]
**Printer firmware version** -  [e.g. 3.8.1, 3.8.1-RC1, ...]
v3.13.2
**MMU upgrade** - [e.g. MMU2S, MMU2, MMU1]
**MMU upgrade firmware version** - [e.g. 1.0.6, 1.0.6-RC2, ...]
MMU3 hardware running 3.0.1
**SD card or USB/Octoprint**
  Please let us know if you print via SD card or USB/Octoprint
SDcard
**Describe the bug**
  A clear and concise description of what the bug is.
When a print pauses due to too many crashes, the extruder cools down.

Then the ONLY option is to review the problem and choose RETRY from the menu, however: it then tries to unload from a cold extruder, which always fails for opvious reasons.  
**To Reproduce**
  Please describe steps to reproduce the behavior.
Make the printer pause due to extruder crashes (.e.g. hit the moving extruder), until the extruder will no longer home and pause. Wait for cool down, then try to recover.  
**Expected behavior**
  A clear and concise description of what you expected to happen.
After selecting "retry" the extruder should be heated up to working temperature, for an unload to succeed.
**G-code**
  Please attach a G-code. This will make it easier for us to replicate the error.
Not relevant.
**Video**
  Please attach a video. It usually helps to solve the problem.
Sorry: not available, let me know if you cannot reproduce from my description.

kevinsaucier, 2024-03-18:
Just piling on here to make sure this one gets some love.  I've had this happen multiple times when a print crashes overnight and I try to resolve it in the morning.  The only fix is to reset the printer and set the nozzle temp before it attempts to resume the unload loop again.  It will then heat up enough and 'usually' allow you to unload, though I still sometimes have to pull the PTFE tube from the extruder and manually pull the filament out.  Less than ideal, for sure.

Majklzbastlirny, 2024-04-02:
Just had the same issue (MK3s+ FW: 3.13.3-7094 [With Revo], MMU3 FW:3.0.2

Patronics, 2024-07-02:
I'm encountering the same issue, it's especially problematic because it grinds the filament, so can cause more major clogs. It's also frustrating that there's no option in the interface to reheat the nozzle first, or to not attempt an unload.

closed 2024-10-16 by github-actions[bot]