prusa3d/Prusa-Firmware #4491: [QUESTION] Bypass 'first startup calibration' message.

hreality, 2023-11-11:
So, I built a bear and we all already know for one reason or another it fails the initial startup calibration.  Ive manually done my calibration and its printing fine but every time Ive turned it on I get the notice the initial calibration wasnt complete and I have to click through it.  What code can I enter into pronterface to trip that as complete so that message goes away?

Thanks

3d-gussner, 2023-11-12:
@hreality Please check with the bear upgrade group which firmware is best to use https://github.com/gregsaun/prusa_i3_bear_upgrade

To overwrite the calibration status is possible but not the preferred way. Please check if there is an community firmware for the bear printer to solve it before you change the EEPROM value.

**_Changing the EEPROM values is possible with D-code. The `D3` is case sensitive and one tiny error may cause a corrupted EEPROM._**

Pronterface, Octoprint and some other serial terminals AUTO UPPERCASE G/D-codes by DEFAULT and this WILL/MAY break/corrupt the EEPROM!
 
See for more information https://github.com/prusa3d/Prusa-Firmware/wiki/EEPROM-table
and look for `EEPROM_CALIBRATION_STATUS_V2`

hreality, 2023-11-18:
Thank you.. I'll give it a look.. seems change to a value of 1 is sufficient

3d-gussner, 2023-11-29:
@hreality As this ticket hasn't been updated I close it. If you still have an issue please reopen it.

closed 2023-11-29 by 3d-gussner

eefweenink, 2024-07-24:
Extra information for who needs it (like I did ;-)) 
I have the same issue. XYZ cal finds I should consult manual. I measured all (x times). Yes there are issues (skew Y-carriage) and I think I can work with it. Every time I bootup, I see the message to do the calibration. And octoprint cannot connect unless I hit the knob on the lcd first. 
So for me best would be to disable this message. 

I have read the above advice & links and found even more of @3d-gussner (very helpfull!)
As I understand I have to set this bit to 1: 
0x0CA6 EEPROM_CALIBRATION_STATUS_V2 02h XYZ cal passed

To do so: 
I opened the terminal on octoprint to read the value first:
-------------
Send: D3 Ax0ca6 C1
Recv: D3 - Read/Write EEPROM
Recv: 0ca6  04
Recv: ok
-----------------
This means (as I understand) that "Z cal passed" is 1 and the others are all 0
Because I also want "XYZ cal passed" to 1. The value should be 06
So I wrote:
-----------
Send: D3 Ax0ca6 X06
Recv: D3 - Read/Write EEPROM
Recv: 1 bytes written to EEPROM at address 0x00000ca6
Recv: 0ca6  06
Recv: ok
Reading it back after reboot:
------------------
Send: D3 Ax0ca6 C1
Recv: D3 - Read/Write EEPROM
Recv: 0ca6  06
Recv: ok

But no success (yet). I still keep getting the same message.
So there must be something else what has to be changed. 
--------
I did the other calibrations: selftest, Z cal (not XYZ) and first layer. And no message anymore

eefweenink, 2024-07-24:
### Extra information for who needs it (like I did ;-))  (and I have no idea how to format this message to simple"
I have the same issue. XYZ cal finds I should consult manual. I measured all (x times). Yes there are issues (skew Y-carriage) and I think I can work with it. Every time I bootup, I see the message to do the calibration. And octoprint cannot connect unless I hit the knob on the lcd first. 
So for me best would be to disable this message. 

I have read the above advice & links and found even more of @3d-gussner (very helpfull!)
As I understand I have to set this bit to 1: 
0x0CA6 EEPROM_CALIBRATION_STATUS_V2 02h XYZ cal passed

To do so: 
I opened the terminal on octoprint to read the value first:
-------------
Send: D3 Ax0ca6 C1
Recv: D3 - Read/Write EEPROM
Recv: 0ca6  04
Recv: ok
-----------------
This means (as I understand) that "Z cal passed" is 1 and the others are all 0
Because I also want "XYZ cal passed" to 1. The value should be 06
So I wrote:
-----------
Send: D3 Ax0ca6 X06
Recv: D3 - Read/Write EEPROM
Recv: 1 bytes written to EEPROM at address 0x00000ca6
Recv: 0ca6  06
Recv: ok
Reading it back after reboot:
------------------
Send: D3 Ax0ca6 C1
Recv: D3 - Read/Write EEPROM
Recv: 0ca6  06
Recv: ok

But no success (yet). I still keep getting the same message.
So there must be something else what has to be changed. 
--------
I did the other calibrations: selftest, Z cal (not XYZ) and first layer. And no message anymore

3d-gussner, 2024-07-24:
The EEPROM_CALIBRATION_STATUS_V2 is a using bits to set the state of calibration
- 01 = Selftest passed
- 02 = XYZ cal passed
- 03 = Selftest and XYZ cal passed
- 04 = Z cal passed
- 05 = Selftest and Z cal passed
- 06 = XYZ cal and Z cal passed
- 07 = Selftest, XYZ cal and Z cal passed
- 08 = Thermal Model cal passed
- 09 = Selftest and Thermal Model cal passed
- 0a = XYZ cal and Thermal Model cal passed
- 0b = Selftest, XYZ cal and Thermal Model cal passed
- 0c = Z cal and Thermal Model cal passed
- 0d = Selftest, Z cal and Thermal Model cal passed
- 0e = XYZ cal, Z cal and Thermal Model cal passed
- 0f = Selftest, XYZ cal, Z cal and Thermal Model cal passed
- 10 = Live Adjust passed
...
- 1f = Selftest, XYZ cal, Z cal, Thermal Model cal and Live Adjust passed

Hope that helps.

PyMeH, 2025-04-21:
Reboot seems to wipe out my changes.
Before reboot:

```
Send: D3 Ax0ca6 X1f
Recv: D3 - Read/Write EEPROM
Recv: 1 bytes written to EEPROM at address 0x0ca6
Recv: 000ca6  1f
Recv: ok
[...]
Send: D3 Ax0ca6 C6
Recv: D3 - Read/Write EEPROM
Recv: 000ca6  1f 01 00 00 00 00

```

In other words - address 0x0AC6 is updated with value 0x1F (calibrations passed)
(by the way I compiled the eeprom.h and verified that indeed EEPROM_CALIBRATION_STATUS_V2 == 0x0AC6. so the address is correct.)

Then I click the reset button of the printer and:

```
Send: D3 Ax0ca6 C6
Recv: D3 - Read/Write EEPROM
Recv: 000ca6  00 01 00 00 00 00

```

Only me?

PyMeH, 2025-04-21:
> Reboot seems to wipe out my changes. The OctoPrint gcode console before reboot:
>  .........
> Only me?

Damn!!
Reset seems to wipe my changes.
Power off and power-on keeps them!!!
So, I've spent I-dont-know-how-many hours building firmware and trying all kinds of crazy ideas ... just to realise that my habit of rebooting the printer with the Reset button is the root of all evil!

3d-gussner, 2025-04-22:
@PyMeH Thanks for sharing the information and glad you got it solved.