prusa3d/Prusa-Firmware #4492: [ENHANCEMENT] Add support for 0.2mm and 1.2mm nozzle sizes

nazar-pc, 2023-11-11:
Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - XL, but likely all others

**Is your enhancement related to a problem? Please describe.**
Currently nozzle size is limited to 0.25-1.0mm range, while I have 0.2mm and 1.2mm nozzles that I'd like to try. I was able to print successfully with 0.2mm nozzle while configuring 0.25mm on the printer, but it complains that gcode doesn't match installed nozzle size.
  
**Describe the solution you'd like**
I'd like to see nozzle size range increased in future firmware versions, probably to 0.2-1.2mm range.

nazar-pc, 2023-11-11:
I just realized this should have been posted at https://github.com/prusa3d/Prusa-Firmware-Buddy instead, can someone move it, please?

3d-gussner, 2023-11-12:
@nazar-pc I think it will be faster if you create a new issue and just copy paste the text above.

Please also fill in the template https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/new?assignees=&labels=enhancement&projects=&template=enhancement.md&title=%5BENHANCEMENT%5D+%3CEnter+comprehensive+title%3E with all asked information.

closed 2023-11-12 by 3d-gussner