prusa3d/Prusa-Firmware #4496: Firmware 3.13.2 not works with MMU"S(MK2.5S +MMU2S)

HP-Munich, 2023-11-12:
**Printer type** - [e.g. MK3S, MK3, MK2.5S, MK2.5, MK2S, MK2]
**Printer firmware version** -  [e.g. 3.8.1, 3.8.1-RC1, ...]

**MMU upgrade** - [e.g. MMU2S, MMU2, MMU1]
**MMU upgrade firmware version** - [e.g. 1.0.6, 1.0.6-RC2, ...]

**SD card or USB/Octoprint**
  Please let us know if you print via SD card or USB/Octoprint

**Describe the bug**
  A clear and concise description of what the bug is.
  
**To Reproduce**
  Please describe steps to reproduce the behavior.
  
**Expected behavior**
  A clear and concise description of what you expected to happen.
  
**G-code**
  Please attach a G-code. This will make it easier for us to replicate the error.

**Video**
  Please attach a video. It usually helps to solve the problem.

HP-Munich, 2023-11-12:
The Firmware not works with MMU2S ( MK2.5S + MMU2S)

HP-Munich, 2023-11-12:
he Firmware 3.13.2 not works with MMU2S ( MK2.5S + MMU2S)

3d-gussner, 2023-11-12:
@HP-Munich Which firmware do you have on MMU2S ? Please also fill in the Issue template correctly.
Please read https://help.prusa3d.com/guide/10b-mk3s-setup-and-calibration_334155#335698

3d-gussner, 2023-11-29:
Closing as no response after two weeks

closed 2023-11-29 by 3d-gussner