prusa3d/Prusa-Firmware #4497: [BUG] MK3S 3.12.2 " model was sliced for another printer model"

Asrothear, 2023-11-12:
**Printer type** - [e.g. MK3S]
**Printer firmware version** -  [3.12.2]

**MMU upgrade** -MMU2S]
**MMU upgrade firmware version** - [1.0.6]

**SD card or USB/Octoprint**
  via SD card

**Describe the bug**
  After flashing the prusa3d_fw_MK3S_MK3S+_3_13_2_7080.hex via Prusa Slicer, the printer shows a message that the model was sliced for another printer model.
  Prusa Slicer version is 2.6.1, configs are up-to-date
  
**To Reproduce**
  Update I2 Mk3S FW? don't know if that's enough to reproduce

**Expected behavior**
  Printer should print without "complains"
  
**G-code**
  Please attach a G-code. This will make it easier for us to replicate the error.
[Test1_0.2mm_PLA_MK3SMMU2S_1h7m.gcode.txt](https://github.com/prusa3d/Prusa-Firmware/files/13328917/Test1_0.2mm_PLA_MK3SMMU2S_1h7m.gcode.txt)

**Video**
  If really needed sure. Right now, I'm in need of the printer and use an older FW ^^

FrHePo, 2023-11-12:
The gcode says: printer_model = MK3SMMU2S but the gcode itself was generated for: print_settings_id = 0.20mm SPEED @MK3
This causes the warning that the model was sliced for another printer model.
AFAIK this is correct behavior and not a bug.

Asrothear, 2023-11-12:
> The gcode says: printer_model = MK3SMMU2S but the gcode itself was generated for: print_settings_id = 0.20mm SPEED @MK3
> This causes the warning that the model was sliced for another printer model.
> AFAIK this is correct behavior and not a bug.

That's in the commented section, the GCode says clearly "M862.3 P "MK3SMMU2S" ; printer model check"

3d-gussner, 2023-11-13:
@Asrothear Thanks for reporting this issue.
After comparing the previous MK3S+MMU2S and new MK3S+MMU3 Prusa Slicer profiles it makes sense not to warn old sliced gcode files containing `MK3SMMU2S` printer names as these are nearly identical.
We plan to fix it in next release.

For the time being as a workaround:

Replace in the old gcode files on your SD card the `M862.3 P "MK3SMMU2S"` with `M862.3 P "MK3SMMU3"` to avoid the warning.

Please use a good text editor like notepad++, vscode... The Windows notepad may introduce some issue while editing the gcode files.

Asrothear, 2023-11-13:
Thanks for the reply.
Would it be possible to flash the mmu3 fw onto the muu2 or are there some major changes in hardware and electronics?

> @Asrothear Thanks for reporting this issue. After comparing the previous MK3S+MMU2S and new MK3S+MMU3 Prusa Slicer profiles it makes sense not to warn old sliced gcode files containing `MK3SMMU2S` printer names as these PrusaSlicer profiles are nearly identical. We plan to fix it in next release.
> 
> For the time being as a workaround:
> 
> Replace in the old gcode files on your SD card the `M862.3 P "MK3SMMU2S"` with `M862.3 P "MK3SMMU3"` to avoid the warning.
> 
> Please use a good text editor like notepad++, vscode... The Windows notepad may introduce some issue while editing the gcode files.

Asrothear, 2023-11-13:
And just to be sure, im using a i2 Mk3S with a MMU2S.
So that's not an "old" sliced model.
[image]

Asrothear, 2023-11-13:
So after Flashing the 3.12.2 FW the I3 Mk3S stops recognizing the MMU2, either i use the MMU2 or MMU3 Fw

3d-gussner, 2023-11-13:
@Asrothear The FW3.13.2 only supports MMU FW https://github.com/prusa3d/Prusa-Firmware/releases/download/v3.13.2/MMU3_3.0.1+837.hex.
See https://help.prusa3d.com/article/mmu3-firmware-compatibility_470808

The new MMU firmware works "best" with the MMU3 hardware https://www.prusa3d.com/product/original-prusa-mmu2s-to-mmu3-upgrade-for-mk3s-2/
The MMU firmware 3.0.1 on MMU2S is working but there is a reason why we changed few thing on the hardware side.

Asrothear, 2023-11-13:
> @Asrothear The FW3.13.2 only supports MMU FW https://github.com/prusa3d/Prusa-Firmware/releases/download/v3.13.2/MMU3_3.0.1+837.hex.
> See [...]

As mentioned, the MMU3 FW is not recognized by the I3 MK3 on 3.12.2
That's seems to be the problem, the GCode reports the usage if the MMU2 but the printer don't detect the MMU2. So the printer shows the prompt "the model was sliced for another printer model"

3d-gussner, 2023-11-14:
@Asrothear You need to flash https://github.com/prusa3d/Prusa-Firmware/releases/download/v3.13.2/MMU3_3.0.1+837.hex to the MMU with PrusaSlicer using the Micro USB cable.
The MMU is disabled by default and has to be enabled in Settings -> MMU [On|Off]

Please slice a new MMU print choosing the MK3S + MMU3 profile. You may need to run the PrusaSlcier Wizard to add the MK3 + MMU3.

Please connect your printer via USB and start a logging using Putty see [here](https://help.prusa3d.com/article/crash-dump_364959) how to configure it.

Send `M115` after connecting with Putty to your printer
Turn on MMU in Settings

Report back the log file while it starts up

Asrothear, 2023-11-18:
Seems somewhat inconvenient to "default" deactivate the MMU in the new FW, without any hints.
That was the problem, and i just didn't checked it

3d-gussner, 2023-11-21:
@Asrothear

> Seems somewhat inconvenient to "default" deactivate the MMU in the new FW, without any hints.
That was the problem, and i just didn't checked it

We have mention that the MMU is disabled in https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.13.0 and we need to keep the release notes as short as possible (otherwise even less people would read it).

Sorry for the inconvenience.

3d-gussner, 2023-11-21:
We also decided NOT to change the M862.3 gcode check. At this moment the MK3S+MMU3 Prusa Slicer profile is more or less the same as the MK3S+MMU2S but this may change and then the message is NEEDED to inform the user.
The workaround for the "old" gcode I described here https://github.com/prusa3d/Prusa-Firmware/issues/4497#issuecomment-1807712490

Please consider to close this issue.

closed 2023-11-21 by Asrothear