prusa3d/Prusa-Firmware #4498: [BUG]<Tune/change filament from LCD-menu causes misalignment>

FrHePo, 2023-11-12:
Printer type: MK3S
Printer firmware version: 3.13.2
print via SD card

When changing filament during printing with the LCD-function <Tune/change filament> the print head doesn't return to the correct position creating a X-misalignment of about 0.7mm.
After I noticed this I did a cube test print (see picture) which confirmed me that this misalignment is independent of the printed object or the G-code.
[image]
[image]

cozmo14047, 2023-11-14:
I have this but it's not as bad.

3d-gussner, 2023-11-15:
@FrHePo @cozmo14047 Thanks for the bug report.
I added few extra code lines to v3.13.2 to trouble shoot this and can't find any layer shift during the M600 color change.

Below part of the serial output:
- At the 1st line the M600 is called and the printer waits to finish the movement until line 3
- Line 3 is the position the printer stopped printing
- Line 4 and 5 are the pause positions and unload/load filament
- Line 6+7 are the reported positions just before it continues to extrude

As line 3 and line 6 are identical there should be no layer shift caused by the printer firmware.
  
```
1 2023-11-15 11:59:38,077 - Recv: X:139.88 Y:117.50 Z:0.20 E:5.16 Count X: 118.86 Y:120.25 Z:0.29 E:2.39
2 2023-11-15 11:59:38,083 - Recv: X:139.88 Y:117.50 Z:0.20 E:5.16 Count X: 118.66 Y:120.25 Z:0.29 E:2.39
3 2023-11-15 11:59:41,138 - Recv: X:110.32 Y:118.44 Z:0.20 E:6.09 Count X: 110.32 Y:118.44 Z:0.29 E:6.09
4 2023-11-15 11:59:58,225 - Recv: X:211.00 Y:0.00 Z:27.20 E:24.09 Count X: 211.00 Y:0.00 Z:27.47 E:19.11
5 2023-11-15 12:00:00,964 - Recv: X:211.00 Y:0.00 Z:27.20 E:24.09 Count X: 110.32 Y:118.44 Z:27.29 E:24.09
6 2023-11-15 12:00:02,088 - Recv: X:110.32 Y:118.44 Z:0.20 E:6.09 Count X: 110.32 Y:118.44 Z:0.29 E:6.09
7 2023-11-15 12:00:02,095 - Recv: X:110.32 Y:118.44 Z:0.20 E:6.09 Count X: 110.32 Y:118.44 Z:0.29 E:6.09
```

Please try to enable the position auto report with `M155 S1 C4` or `M155 S1 C5` for temperatures and positions.

Tested with MK404 and here a screenshot of a one layer print with a Tune -> Color change
[image]

FrHePo, 2023-11-15:
The misalignment with 0.4 nozzle is the same.

cozmo14047, 2023-11-16:
> @FrHePo @cozmo14047 Thanks for the bug report. I added few extra code lines to v3.13.2 to trouble shoot this and can't find any layer shift during the M600 color change.
>
> Below part of the serial [...]

I am yet to try with this in the gcode color change but it's when I go change fillament via lcd menu.i have it not as bad as @frhepo but still pretty bad I can try the suggested commands

3d-gussner, 2023-11-16:
@FrHePo Thanks for sharing the picture and log file. Let's see if we can find something.

cozmo14047, 2023-11-16:
@3d-gussner I can confirm I don't have the issue when setting color change in gcode I only have it using the lcd menu.

3d-gussner, 2023-11-16:
I have been printing a test cube on real hardware with fw3.13.2 and had no layer shift with lcd tune color change

FrHePo, 2023-11-16:
Well, thanks for testing, but of course I'm still wondering what could cause this. Supposing it's not the firmware what could it be then? Are there any settings that could cause this? I have modded my MK3S allright, but not anything having to do with the X- and Y-axis. My modding concerns the frame (bear) and the extruder.

3d-gussner, 2023-11-17:
@FrHePo I have no idea why that would happen and so repeatable on your printer.
Does that also happen when you inject a `M600` gcode via serial while printing from SD card?
Does that happen when you print a SD print with a `M600` gcode command added within PrusaSlicer?

Do you use the [FW3.13.2+7080 firmware](https://github.com/prusa3d/Prusa-Firmware/releases/download/v3.13.2/prusa3d_fw_MK3_3_13_2_7080.hex) ?

Wasn't there any need to use a custom version for the bear? @gregsaun Just to be sure is there any custom firmware for the bear needed?

cozmo14047, 2023-11-17:
> @FrHePo I have no idea why that would happen and so repeatable on your printer. Does that also happen when you inject a `M600` gcode via serial while printing from SD card? Does that happen when you [...]

So I upgraded to 3.13.2 and now I don't have It so ? Can old firmware plus new slicer equal problems

3d-gussner, 2023-11-17:
@cozmo14047 Glad that the newest firmware solved your issue. Which firmware version did you have before that caused the issue?

Please next time when you react to an issue also add some more information as we ask in the issue templates to avoid chasing an issue that my be related to an older or different version.

FrHePo, 2023-11-17:
> @FrHePo I have no idea why that would happen and so repeatable on your printer. Does that also happen when you inject a `M600` gcode via serial while printing from SD card? Does that happen when you [...]

@3d-gussner Yes, FW3.13.2+7080 is correct. No, there's no custom version of the firmware for the bear frame needed or available. I'll try the manual M600 tonight.

FrHePo, 2023-11-17:
Yep, both ways produce the same misalignment. It's consistent ☹️

3d-gussner, 2023-11-18:
@FrHePo I could prepare a debug firmware version 3.13.2 for you to see if there is anything going on with the firmware or the issue is somewhere else. If you want to try that please send me an email, you find it in my github account.

gudnimg, 2023-11-19:
For what its worth, I'm unable to reproduce the issue on MK3S+

Coordinates after M609 parking is done
* M600 Resume position (mm) - X: 72.761, Y: 53.711, Z: 0.200
* M114 output: X:72.76 Y:53.71 Z:0.20 E:19.95 Count X: 71.51 Y:54.98 Z:0.40 E:19.95

Before unparking is about to start:
* M600 Resume position (mm) - X: 72.761, Y: 53.711, Z: 0.200
* M114 output: X:211.00 Y:-0.00 Z:27.20 E:57.95 Count X: 209.68 Y:1.11 Z:27.22 E:52.97

After unparking is done:
* M114 output: X:72.76 Y:53.71 Z:0.20 E:19.95 Count X: 71.51 Y:54.98 Z:0.40 E:19.95

gudnimg, 2023-11-19:
This may be unrelated but I wonder if your Y-axis is experiencing a "crash"? 🤔 

@FrHePo is crash detection enabled?

On my MK3S+ the crash seems to be completely ignored.

FrHePo, 2023-11-19:
@gudnimg Yes, crash detection is enabled and the position resuming is absolutely smooth.

FrHePo, 2023-11-19:
I just got an idea. I'm using the Revo Micro as my hotend and therefore I also loaded E3DREVO_fw_MK3S_MK3S+_3_13_0_6873.hex. Maybe there's a connection? I'll reload only 3_13_2_7080 and see if it makes any difference.

FrHePo, 2023-11-21:
I changed the firmware back to 3_11_0_4955. The cube was again printed with the same misalignment of 0.7mm.

FrHePo, 2023-11-21:
> @FrHePo I could prepare a debug firmware version 3.13.2 for you to see if there is anything going on with the firmware or the issue is somewhere else. If you want to try that please send me an email, you find it in my github account.

@3d-gussner Did you get my email?

FrHePo, 2023-11-21:
Ok, I think I solved the riddle. Before the old filament can be removed the build-plate travels to the minimum y-position. Because of my changes my minimum y-position must be 0.7mm less than a standard MK3S. Therefore when the build-plate returns to the correct Y-position for the hotend to resume the print the build-plate travels 0.7 mm too far.
For me this behaviour raises 2 questions:
1st: Why doesn't the firmware use the homing data to know the actual min and max Y-values of a printer?
2nd: Is it necessary that the build-plate travels to the extreme Y-position for changing the filament? Couldn't this Y-postion be +2 mm instead of 0mm? Thus leaving some space for tolerance?

3d-gussner, 2023-11-21:
> @3d-gussner Did you get my email?

No I just got github emails.

3d-gussner, 2023-11-21:
> Because of my changes my minimum Y-position must be 0.7mm less than a standard MK3S. Therefore when the build-plate returns to the correct Y-position for the hotend to resume the print the build-plate travels 0.7 mm too far.

Okay at least you have found the issue. Your printer is different than the Bear or Stock printer. 
 

> 1st: Why doesn't the firmware use the homing data to know the actual min and max Y-values of a printer?

> 2nd: Is it necessary that the build-plate travels to the extreme Y-position for changing the filament? Couldn't this Y-postion be +2 mm instead of 0mm? Thus leaving some space for tolerance?

We have defined [FILAMENTCHANGE_[XPOS|YPOS|ZADD] = X211 Y0 Z+2](https://github.com/prusa3d/Prusa-Firmware/blob/v3.13.2/Firmware/variants/MK3S.h#L359-L361)  which are fine for our printers. And normally the Y axis can travel from [Y-4 up to 212.5](https://github.com/prusa3d/Prusa-Firmware/blob/v3.13.2/Firmware/variants/MK3S.h#L64-L65) so for the stock printers it has some tolerances.

Please consider to create an own Firmware/variant/<myprinter>.h file with the modifications and compile the firmware.

3d-gussner, 2023-11-21:
@FrHePo Sorry but I have to remove the BUG label and we can't / won't fix it as all other stock and probably most Bear printers work well.
As this is an open source firmware feel free to modify it to your needs, see my comment above.
  
If we change it other users may/will complain why the filament change position has changed and it is now "harder" to grab the filament when they print in that area.

FrHePo, 2023-11-22:
> > Because of my changes my minimum Y-position must be 0.7mm less than a standard MK3S. Therefore when the build-plate returns to the correct Y-position for the hotend to resume the print the build-plate travels 0.7 mm too far.
> 
> Okay at least you have found the issue. Your printer is different than the Bear or Stock printer.
> 
> > 1st: Why doesn't the firmware use the homing data to know the actual min and max Y-values of a printer?
> 
> > 2nd: Is it necessary that the build-plate travels to the extreme Y-position for changing the filament? Couldn't this Y-postion be +2 mm instead of 0mm? Thus leaving some space for tolerance?
> 
> We have defined [FILAMENTCHANGE_[XPOS|YPOS|ZADD] = X211 Y0 Z+2](https://github.com/prusa3d/Prusa-Firmware/blob/v3.13.2/Firmware/variants/MK3S.h#L359-L361) which are fine for our printers. And normally the Y axis can travel from [Y-4 up to 212.5](https://github.com/prusa3d/Prusa-Firmware/blob/v3.13.2/Firmware/variants/MK3S.h#L64-L65) so for the stock printers it has some tolerances.
> 
> Please consider to create an own Firmware/variant/.h file with the modifications and compile the firmware.

@3d-gussner I still don't understand all of it. I checked the dimensions of my Bear-frame and especially the position of the Z-profiles. All dimensions are correct! If the Bear-frame should cause this misalignment then this should be the case for all people that upgraded to the bear-frame and that will be more than just a few people.

gregsaun, 2023-11-22:
Hello @FrHePo ,

1. Where does your Bear frame come from?
2. Could you check the dimensions as explained in this page: https://github.com/gregsaun/prusa_i3_bear_upgrade/tree/master/optional_parts/bear_2.0_upgrade_mk2-2s-2.5-2.5s#printed-parts ? Do you have a 311 or 331mm length?
3. Have you modified the Y motor mount, Y belt tensioner or y rods holder?

FrHePo, 2023-11-22:
@gregsaun 
1. Lecktor Technologies OÜ - Värvi 5 C400-1 - 10621 Tallinn - Estonia
2. The Y-axis extrusion length is 331mm
3. No, all parts are original Bear.

FrHePo, 2023-11-23:
@gregsaun What do think?

gregsaun, 2023-11-23:
@FrHePo have you tried to disable crash detection?

FrHePo, 2023-11-23:
No. A crash should trigger a LCD-message and there never was any such message. What about moving the Z-profiles some mm further to the back?

gregsaun, 2023-11-23:
It would still be good to test with crash detection disabled, it is a quick test. If it doesn't work could you please open an issue here and copy paste the link of this issue?
https://github.com/gregsaun/prusa_i3_bear_upgrade/issues/

Could you also add some pictures of your printer over there.

Thank you in advance

FrHePo, 2023-11-23:
Ok. I'll do that.

FrHePo, 2023-11-23:
Ok, I disabled the crash detection but the result is exactly the same as before. It's also very clear that when the build-plate travels to the change filament position it runs into the Y-limit (by 0.7mm) which is actually an unintended collision.
If the firmware would use the homing data instead of using a hard coded position I think this problem would not arise. That way the firmware would be more "intelligent".

FrHePo, 2023-11-24:
See also https://github.com/gregsaun/prusa_i3_bear_upgrade/issues/142

3d-gussner, 2023-12-11:
@FrHePo Closing this issue as you have solved it with @gregsaun :hugs: 
If you don't agree to close it feel free to reopen it.

closed 2023-12-11 by 3d-gussner