prusa3d/Prusa-Firmware #4499: [BUG] MK3S 3.13.2 wizard fail for fan error

welverin, 2023-11-13:
after upgrade to 3.13.2 i start the wizard for recalibrate the printer
fan test fail (both fan spinning without problem)
sometimes fail left fan and wizard stopped
after a while the fail is on front fan, wizard ask me if fan spinning, if a click on ok, the wizard continue

3d-gussner, 2023-11-13:
@welverin Thanks for reporting. Please fill in the issue template https://github.com/prusa3d/Prusa-Firmware/issues/new?assignees=&labels=bug&projects=&template=bug_report.md&title=%5BBUG%5D%3CEnter+comprehensive+title%3E
so we don't have to ask if the printer is stock or not.

Are these stock fans?
Do you have a option to connect the printer via USB and monitor / log the wizard?

welverin, 2023-11-14:
> yes, all stock
> yes i can connect the printer via usb but how can i read the log of the wizard ?

welverin, 2023-11-14:
> @welverin Thanks for reporting. Please fill in the issue template [...]

yes, all stock
yes i can connect the printer via usb but how can i read the log of the wizard ?

closed 2023-11-14 by welverin

cnever, 2024-11-24:
What was the solution? I see it's shown as completed but can't find any solution. Thanks