prusa3d/Prusa-Firmware #4500: [BUG]<Enter comprehensive title>

aideym, 2023-11-14:
**Printer type** - Mk3s
**Printer firmware version** - 3.13.2 3.13.1

**MMU upgrade** -na
**MMU upgrade firmware version** -na

**SD card or USB/Octoprint**
   SD card 

**Describe the bug**
TM error is triggered when printer performs the 7x7 test before starting printing at around the 7th point. If I can get it to continue by pressing the knob when asks for supervision then it continues for a few seconds then reports Thermal runaway. Again after intervention it continues to print as per requirements until next model is loaded and then it continues to do exactly the same thing.  
**To Reproduce**
  turn the printer on and try and print a model.
  
**Expected behavior**
to print without requiring intervention on my behalf
  
**G-code**
Connecting...
start
Printer is now online.
;S
;S
;S
;S
;S
SN update failed
echo: 3.13.2-7080
SpoolJoin is Off
echo: Last Updated: Oct 20 2023 17:49:53 | Author: (none, default config)
echo: Free Memory: 2513  PlannerBufferBytes: 1760
echo:Stored settings retrieved
adc_init
Hotend fan type:
NOCTUA
CrashDetect ENABLED!
echo:SD init fail
Received command dump_available
// action:dump_available
echo:Unknown command: "M105"(2)
echo:SD init fail
tmc2130_home_enter(axes_mask=0x04)
echo:Hold current truncated to Run current
tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x04
Sending 0xFF
echo:SD card ok
LCD status changed
echo:Now fresh file: ultima~1.gco
File opened: /Ultimate 40 mm Calibration Test Cube v3_0.2mm_PETG_M Size: 6625651
File selected
LCD status changed
LCD status changed
echo:enqueing "M23 ultima~1.gco"
echo:enqueing "M24"
echo:Now fresh file: ultima~1.gco
File opened: /Ultimate 40 mm Calibration Test Cube v3_0.2mm_PETG_M Size: 6625651
File selected
LCD status changed
NORMAL MODE: Percent done: 0; print time remaining in mins: 146; Change in mins: -1
SILENT MODE: Percent done: -1; print time remaining in mins: -1; Change in mins: -1
NORMAL MODE: Percent done: 0; print time remaining in mins: 146; Change in mins: -1
SILENT MODE: Percent done: 0; print time remaining in mins: 146; Change in mins: -1
LCD status changed
LCD status changed
LCD status changed
LCD status changed
tmc2130_home_enter(axes_mask=0x01)
echo:Hold current truncated to Run current
  0 step=61 mscnt= 989
tmc2130_goto_step 0 61 2 1000
[... 200 lines cut ...]
echo:Move aborted
echo:enqueing "CRASH_RECOVER"
tmc2130_home_enter(axes_mask=0x01)
  0 step=62 mscnt= 994
tmc2130_goto_step 0 61 2 1000
tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x01
tmc2130_home_enter(axes_mask=0x02)
  0 step=17 mscnt= 287
tmc2130_goto_step 1 17 2 1000
tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x02
TM: error |-1.352419|>1.200000
LCD status changed
TM: error |-1.352419|>1.200000
TM: error |-1.260103|>1.200000
TM: error |-1.480193|>1.200000
TM: error |-1.480193|>1.200000
LCD status changed
LA10C: Linear Advance mode: 1.5
echo:Advance K=0.08
K out of allowed range!
echo:Advance K=0.08
NORMAL MODE: Percent done: 0; print time remaining in mins: 145; Change in mins: -1
SILENT MODE: Percent done: 0; print time remaining in mins: 146; Change in mins: -1

I don't currently have video, I'll take one after this print finishes

3d-gussner, 2023-11-14:
@aideym Is your printer stock or a modified one? If modified which mods do you have? Did you run Calibration -> Thermal Model cal.?

Please add `M503` gcode serial out put

3d-gussner, 2023-11-15:
@aideym Please bit more specific which hotend, heater, thermistor etc. Searching for "dragonfly" shows lot of different types.

The thermal model is tailored around the E3Dv6 style hotend Prusa uses.

Any other 3rd party hotend gonna need a new set of values as it behaves differently. Please contact Phaetus.

aideym, 2023-11-15:
OK, thank you.
It's using genuine prusa thermistors but I take your point. I just don't
get why it worked with all the other firmware upgrades but not the last
two. I'll just leave it on the firmware that works and wait for my mk4 .
Thank you
Aiden

3d-gussner, 2023-11-15:
@aideym The thermal model worked for you with the previous version 3.12.0 with the dragon fly?

aideym, 2023-11-15:
Yes it works fine, no problems, I've gone back to it and I'm printing now

3d-gussner, 2023-11-15:
@aideym Which FW is working and which is not?

aideym, 2023-11-15:
prusa3d_fw_MK3S_MK3S+_3_13_0_6873.hex works
prusa3d_fw_MK3S_MK3S+_3_13_1_6876.hex does not work
prusa3d_fw_MK3S_MK3S+_3_13_2_7080.hex does not work

regards Aiden

3d-gussner, 2023-11-15:
@aideym That is odd because the changes from 3.13.0 to 3.13.1 are minimal. And with FW 3.13.2 we introduced the thermal model which probably needs a different version for your dragonfly hotend.

closed 2024-01-23 by github-actions[bot]