prusa3d/Prusa-Firmware #4503: [BUG] TMC error points to non-existent error page (MMU2)

PxT, 2023-11-18:
**Printer type** - MK3S
**Printer firmware version** -  3.13.2

**MMU upgrade** - MMU2S
**MMU upgrade firmware version** - 3.0.1

**SD card or USB/Octoprint**
SD

**Describe the bug**
Received a TMC Driver Error message during a print that points to prusa.io/04321 however that page does not exist and just redirects to the main prusa3d site.

I was able to find the [correct page](https://help.prusa3d.com/article/tmc-driver-error-04321-mmu_492957) with Google but that text only mentions MMU3 and not the MMU2 on my machine.

Prusa-Support, 2023-12-16:
Thanks for reporting.

This is not on the firmware side but a routing error across our websites.
This error has been fixed now.

Michele Moramarco
Prusa Research

closed 2023-12-16 by Prusa-Support

duebelpapst, 2024-01-11:
Hi PxT,

do you find a solution for the problem?

Regards,
Stefan

PxT, 2024-01-11:
@duebelpapst  Installing the PD board solved all the power issues and driver errors for me

duebelpapst, 2024-01-11:
@PxT From were do you get the PD Board, I can't find it at the Prusa shop.

PxT, 2024-01-11:
@duebelpapst This isn't really how GitHub is supposed to be used so I'm not going to link to a 3rd party site, but a quick google for "prusa PD board" should bring up a source. Prusa does not sell this part separately for some reason.

duebelpapst, 2024-01-11:
@PxT  Thanks, I had already found it, just thought you had managed to buy the part from Prusa.
At the moment the MMU2S is running after I increased the voltage on the power supply from 23.9V to 24.5V.
But in the future I will supply the MMU2S board directly with a step down converter with a little more than 5V, which I will connect to the 24V of the borad.