prusa3d/Prusa-Firmware #4505: > I have tried 3.13.0, .1 and .2, all 3 of them give me issues with thermal anomalies and eventually thermal errors. I've done the steps above, i've factory reset, manually PID tune, etc. and nothing but thermal errors (and lots of beeping).

JWartenberg, 2023-11-20:
In my case it was `E3DREVO_fw_MK3S_MK3S+_3_13_2_7080.hex`. The default download will not work. You need to download from Github and use the correct version (including heater power).

3d-gussner, 2023-11-20:
@JWartenberg Please read https://github.com/prusa3d/Prusa-Firmware/issues/4451 and contact E3D support

laure3d, 2023-11-27:
I had the same issue, I spent the whole afternoon with the Prusa help, they made me run tests, calibrations etc and finally they made me flash back 3.11.0 firmware and it worked but they said it was only a temporary solution and they are sending me a new thermistor and a new hotend. I don't know what they did with this update

I99Developer, 2023-12-30:
Rolled back to 3.12.2 and the problem is gone.  I tried 13.3.1 and up with no luck.

JWartenberg, 2023-12-30:
After endless thermal errors I decided to turn off the feature.
After 2-3 weeks of happy printing I got a termal runaway error. Cold start, calibration didn't help. I checked the cables and found that the thermistor cable on the board was a bit loose.
I turned the feature back on, run the calibration and have been printing since then without any thermal errors.
My impression: if you get a termal anomaly error, you really have a problem or it is an early warning that you will have a problem. But it may be difficult to pinpoint the problem.

3d-gussner, 2023-12-30:
@JWartenberg Thanks for the feedback and highlighting that you should investigate and check your printer and not just turn it off when you get a Thermal Model warning or error. There is a reason why it triggers.

Glad you found the "loose" connection and keep the TM on. 

> But it may be difficult to pinpoint the problem.

It isn't easy to point out the causing component as there are many parameters that can trigger the TM errors.
 
Here my approach off a check list / troubleshooting 
1. Check connections and cables
   1. Thermistor and heater cables
      a. Connection on controller board
      b. Wires intact at the thermistor and hotend
      c. Set some temperature and wiggle the the cable loom for potential (semi)broken cables while watching the temp on LCD or serial output `M155 S1 C1`
      d. Move hotend X min to max and back
      e. Move the hotend in X and Z
   2. Check that the heater and thermistor aren't loose in heater block
 2.  Check part cooling fan
     1. Turn on part fan to 100% and repeat the steps `1.i.c` to `1.i.e` while watching the RPM in support menu or serial output `M155 S1 C3`
     2. Check fan shroud is tight and not moving
3. Some models may cause a blow back of the part cooling fan direction the heater block/nozzle. This also can be challenging so try to print something simple to ensure that it isn't a permanent issue but model related.
   1. Hotend socks can help here, but the stock E3D socks may cause other issues as they may partially block off fall off it is not recommend by Prusa to use them.
   2. Sometimes rotating the model also reduces TM errors on that kind of models.

laure3d, 2023-12-30:
I had the same issue. I managed to print going back to 3.11.1 as the Prusa chat advised me to but they did warn me it’d be temporary and indeed it was. After a few weeks though I could still print, the nozzle temperature was not stable, going from 205 to 230 all of a sudden. Prusa sent me a new thermistor and a new hotend. I only changed the thermistor, went back to the latest firmware and it worked. Now my printer is as good as new!

3d-gussner, 2023-12-30:
@laure3d Thanks for your feedback, glad to hear that the TM pointed out the issue and you fixed the printer with a replacement part.
Nice to see some positive feedback from the community about the Thermal Model.  :hugs:

3d-gussner, 2024-03-08:
Closing as inactive

closed 2024-03-08 by 3d-gussner